zephyrproject-rtos/zephyr #24701: tests/lib/cmsis_dsp/transform failed on frdm_k64f board.

chen-png, 2020-04-26:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f -p auto tests/lib/cmsis_dsp/transform
    -DCONF_FILE=prj_base.conf
    -DCONFIG_CMSIS_DSP_TEST_TRANSFORM_RQ15=y

2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-1974-g4d7d3a25c127  ***
Running test suite transform_rq15

starting test - test_arm_rfft_q15_noisy_32
PASS - test_arm_rfft_q15_noisy_32

starting test - test_arm_rifft_q15_noisy_32
**Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/transform/src/rq15.c:145: test_arm_rifft_q15: (test_snr_error_q15(length, output, ref, SNR_ERROR_THRESH_IFFT) is false)
signal-to-noise ratio error limit exceeded**
FAIL - test_arm_rifft_q15_noisy_32

starting test - test_arm_rfft_q15_noisy_64
PASS - test_arm_rfft_q15_noisy_64

starting test - test_arm_rifft_q15_noisy_64
PASS - test_arm_rifft_q15_noisy_64

starting test - test_arm_rfft_q15_noisy_128
PASS - test_arm_rfft_q15_noisy_128

starting test - test_arm_rifft_q15_noisy_128
**Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/transform/src/rq15.c:145: test_arm_rifft_q15: (test_snr_error_q15(length, output, ref, SNR_ERROR_THRESH_IFFT) is false)
signal-to-noise ratio error limit exceeded**
FAIL - test_arm_rifft_q15_noisy_128

starting test - test_arm_rfft_q15_noisy_256
PASS - test_arm_rfft_q15_noisy_256

starting test - test_arm_rfft_q15_noisy_512
PASS - test_arm_rfft_q15_noisy_512

starting test - test_arm_rfft_q15_noisy_1024
PASS - test_arm_rfft_q15_noisy_1024
....
starting test - test_arm_rfft_q15_step_2048
PASS - test_arm_rfft_q15_step_2048

starting test - test_arm_rfft_q15_step_4096
PASS - test_arm_rfft_q15_step_4096

Test suite transform_rq15 failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.2
 - Commit ID: 4d7d3a25c1

stephanosio, 2020-04-29:
Note: `CONFIG_CMSIS_DSP_TEST_TRANSFORM_RQ31` fails also.

stephanosio, 2020-04-29:
This is a bug (or at least an undocumented/very questionable behaviour) in the CMSIS-DSP inverse FFT implementation; I have filed an upstream issue for this: https://github.com/ARM-software/CMSIS_5/issues/906

For now, I will implement workarounds in the tests to allow predictable operation.

PR #24809 (2020-05-04): tests: lib: cmsis_dsp: transform: Provide RIFFT bug workaround
This commit provides the workarounds for the CMSIS-DSP RIFFT input
buffer access bug reported in #24701.

The upstream issue for this bug is ARM-software/CMSIS_5#906.

Fixes #24701.
closed 2020-05-04 by MaureenHelm