zephyrproject-rtos/zephyr #24702: tests/drivers/counter/counter_basic_api failed on frdm_k64f board.

chen-png, 2020-04-26:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f -p auto tests/drivers/counter/counter_basic_api
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-1974-g4d7d3a25c127  ***
Running test suite test_counter

starting test - test_set_top_value_with_alarm
E: Wrap can only be set to 0xffffffff.
Skipped for RTC_0
PASS - test_set_top_value_with_alarm

starting test - test_single_shot_alarm_notop
Testing RTC_0
D: Current time is 0 ticks
D: Alarm set to 2 ticks
D: Current time is 3 ticks
**Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:279: alarm_handler: (diff < counter_us_to_ticks(dev, processing_limit_us) is false)
Unexpected distance between reported alarm value(3) and actual counter value (3), top:-1 (processing time limit (0 us) might be exceeded?**
FAIL - test_single_shot_alarm_notop

starting test - test_single_shot_alarm_top
E: Wrap can only be set to 0xffffffff.
Skipped for RTC_0
PASS - test_single_shot_alarm_top

starting test - test_multiple_alarms
Skipped for RTC_0
PASS - test_multiple_alarms

starting test - test_all_channels
Testing RTC_0
D: Current time is 0 ticks
D: Alarm set to 2 ticks
D: Current time is 3 ticks
PASS - test_all_channels

starting test - test_late_alarm
Skipped for RTC_0
PASS - test_late_alarm

starting test - test_late_alarm_error
Skipped for RTC_0
PASS - test_late_alarm_error

starting test - test_short_relative_alarm
D: Current time is 0 ticks
D: Alarm set to 1 ticks
D: Current time is 2 ticks
**Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:279: alarm_handler: (diff < counter_us_to_ticks(dev, processing_limit_us) is false)
Unexpected distance between reported alarm value(2) and actual counter value (2), top:-1 (processing time limit (0 us) might be exceeded?**
FAIL - test_short_relative_alarm

starting test - test_cancelled_alarm_does_not_expire
Skipped for RTC_0
PASS - test_cancelled_alarm_does_not_expire

starting test - test_set_top_value_without_alarm
E: Wrap can only be set to 0xffffffff.
Skipped for RTC_0
PASS - test_set_top_value_without_alarm

Test suite test_counter failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain : zephyr-sdk-0.11.2
 - Commit ID: 4d7d3a25c1

pabigot, 2020-05-11:
@MaureenHelm the bug here appears to be this:
```
                        clock-frequency = <32768>;
                        prescaler = <32768>;
```
in the `nxp,kinetis-rtc` devicetree sources for k6x, k8x, and ke1xf, combined with this code in the corresponding driver:
```
		.freq = DT_INST_PROP(0, clock_frequency) /
				DT_INST_PROP(0, prescaler),
```
which causes the RTC to claim its frequency is 1 Hz.  All tested delays convert from microseconds via `counter_us_to_ticks() to zero ticks, and the measured delta zero is never less than zero.

While the test arguable be `diff <= limit` that passes, but it takes a very long time to do so and the conversion is still clearly wrong.

Hope this is somewhat helpful but I don't know what the proper fix is; changing the prescaler to 1 causes other parts of the test to fail.  (The prescaler doesn't seem to be referenced elsewhere in the driver which is also odd.)

MaureenHelm, 2020-05-18:
> which causes the RTC to claim its frequency is 1 Hz.

The RTC frequency is indeed 1 Hz.

Fixed by #25364

closed 2020-05-18 by MaureenHelm