zephyrproject-rtos/zephyr #24703: hal_nuvoton: Add new module for Nuvoton numicro M480 HAL layer

ssekar15, 2020-04-26:
## Introduction

Support for Nuvoton numicro M480 series ARM cortex M4 SoC needs/depends on M480 BSP. New HAL module needs to be created to address the dependency.

### Problem description

I am submitting the patch to add support for Nuvoton M480 SoC in PR. This PR needs HAL layer for Nuvoton numicro M480 BSP.

### Proposed change

hal_nuvoton is currently pushed with needed adaptation for Zephyr here: https://github.com/linumiz/hal_nuvoton

This enables the HAL support for all Numicro M48X series SoC's from Nuvoton

## Detailed RFC

HAL layer can reviewed from here: https://github.com/linumiz/hal_nuvoton

### Proposed change (Detailed)

NA

### Dependencies

NA

### Concerns and Unresolved Questions

This hal layer is taken from source https://github.com/OpenNuvoton/M480BSP, based on release v3.05.000

## Alternatives

NA

ssekar15, 2020-04-26:
@carlescufi @nashif

carlescufi, 2020-04-26:
@ssekar15 thank you for the proposal. Can you commit to maintaining this platform if we create the repo?

ssekar15, 2020-04-26:
@carlescufi yes, I will be able to maintain it. Thanks

carlescufi, 2020-04-27:
> @carlescufi yes, I will be able to maintain it. Thanks

Thank you.

A couple more questions from browsing https://github.com/OpenNuvoton/M480BSP:

- I can't seem to find the license the code is released under
- There seem to be binary blobs in the repository

stephanosio, 2020-04-27:
Found at least one reference to GPL ...
https://github.com/linumiz/hal_nuvoton/blob/a7a6f1e5635d26aea82041b6969683845f5a5fa0/m48x/Devices/M480/Source/GCC/_syscalls.c#L3

ssekar15, 2020-04-27:
@carlescufi Currently license is not available in Nuvoton M480BSP, we have raised issue and Nuvoton is working on legal side to get free licensing. More information can be find in below

https://github.com/OpenNuvoton/M480BSP/issues/9

carlescufi, 2020-04-27:
> @carlescufi Currently license is not available in Nuvoton M480BSP, we have raised issue and Nuvoton is working on legal side to get free licensing. More information can be find in below
> 
> [OpenNuvoton/M480BSP#9](https://github.com/OpenNuvoton/M480BSP/issues/9)

Thank you for the info. About the binary blobs (libraries in binary format), you will need to remove those before you add the code to the Zephyr repo. We do not accept precompiled code right now.

ssekar15, 2020-04-27:
We have extracted only the drivers and device files from Nuvoton M480BSP to zephyr compatible which doesn't have any binary blobs. Could you please confirm are you referring to below or did I misunderstood?
https://github.com/linumiz/hal_nuvoton

carlescufi, 2020-04-27:
> We have extracted only the drivers and device files from Nuvoton M480BSP to zephyr compatible which doesn't have any binary blobs. Could you please confirm are you referring to below or did I misunderstood?
> https://github.com/linumiz/hal_nuvoton

Yep, this looks fine, sorry. I was looking at https://github.com/OpenNuvoton/M480BSP

carlescufi, 2020-05-06:
@ssekar15 sorry for the delay, I've been busy with the release. I should be able to create it tomorrow.

carlescufi, 2020-05-06:
@ssekar15 could you please point us to the hardware documentation for this hardware for the record, so we know we can fix issues by looking at a datasheet or reference manual?

stephanosio, 2020-05-06:
> @ssekar15 sorry for the delay, I've been busy with the release. I should be able to create it tomorrow.

Just a reminder, the licensing terms have not been clarified by Nuvoton yet (https://github.com/OpenNuvoton/M480BSP/issues/9).

parthitce, 2020-05-06:
> @ssekar15 could you please point us to the hardware documentation for this hardware for the record, so we know we can fix issues by looking at a datasheet or reference manual?

TRM: https://www.nuvoton.com/export/resource-files/TRM_M480_Series_EN_Rev2.02.pdf
Datasheet: https://www.nuvoton.com/export/resource-files/DS_M480_Series_EN_Rev2.02.pdf

carlescufi, 2020-05-07:
> > @ssekar15 sorry for the delay, I've been busy with the release. I should be able to create it tomorrow.
> 
> Just a reminder, the licensing terms have not been clarified by Nuvoton yet ([OpenNuvoton/M480BSP#9](https://github.com/OpenNuvoton/M480BSP/issues/9)).

Indeed, I will wait until those are settled since right now we don't have a valid license right now. @stephanosio thanks for bringing this up to my attention.

carlescufi, 2020-05-07:
@parthitce thank you for the TRM and datasheet links. Please do ping me here as soon as the license issue has been resolved.

ssekar15, 2020-05-14:
@carlescufi Nuvoton has changed the valid license Apache 2.0 in the Nuvoton M480 BSP and same has adapted to hal_nuvoton https://github.com/linumiz/hal_nuvoton.

Could you please check and provide feedback if anything else is blocking for repository creation?

carlescufi, 2020-05-27:
> @carlescufi Nuvoton has changed the valid license Apache 2.0 in the Nuvoton M480 BSP and same has adapted to hal_nuvoton https://github.com/linumiz/hal_nuvoton.
> 
> Could you please check and provide feedback if anything else is blocking for repository creation?

@ssekar15 the only thing blocking it is a wider check among different maintainers: @galak @nashif @MaureenHelm @ioannisg please comment. Unless any objections are raised I will create the repository. We are currently working towards a better process for creating and maintaining Zephyr modules but since this was opened long before that discussion started it is my opinion that we should create this repository now.

ssekar15, 2020-06-05:
@carlescufi @galak @MaureenHelm @nashif 
Any update on this please !

jettr, 2020-06-16:
@ssekar15 we are working with Nuvoton to add their chip support for the npcx7 and npcx9 series chips to the main Zephyr tree. We should align or at least see how the different chip support code will interact with each other.

ssekar15, 2020-06-16:
@carlescufi @jettr We have plan to cover other chip series as well, nevertheless I feel it doesn't block to merge M48X which is already ready.

We shall create separate slack channel for Nuvoton to discuss.

jettr, 2020-06-17:
We don't need to block. I just want to make sure that there is thought in the larger picture of Nuvoton support in Zephyr. We can always move files around later if needed.

Please do add me to the slack channel if you create one.

Also, would it be possible to add a root LICENSE file that governed the entire module?

ssekar15, 2020-06-18:
@jettr Thanks for clearification.

Regarding LICENSE file, we request with Nuvoton but  they cannot share since HAL layer has other libraries (other license). So we have to create one by ourself which I believe neither good solution or nor mandatory.

@carlescufi any feedback to proceed?

carlescufi, 2020-06-22:
@ssekar15 finally done:
https://github.com/zephyrproject-rtos/hal_nuvoton

closed 2020-06-22 by carlescufi

ssekar15, 2020-06-22:
@carlescufi Thanks :)