zephyrproject-rtos/zephyr #24709: Counter function with ARM CMSDK Dual Timer

kkiyota, 2020-04-26:
Hi, 

I'm using ARM CMSDK Dual Timer in my custom board with Cortex-M7.  "counter_set_channel_alarm" returns -ENOTSUP.

> int counter_set_channel_alarm(struct device *dev, u8_t chan_id, const struct counter_alarm_cfg *alarm_cfg)

Looking at freya/zephyr/drivers/counter/timer_dtmr_cmsdk_apb.c, there seems to be no corresponding function defined.

Is it intentional?   Or future improvement?

Thank you for your help!

kkiyota, 2020-04-28:
Looking at zephyr/drivers/counter/timer_dtmr_cmsdk_apb.c, the following APIs are only supported.

counter_set_channel_alarm is not supported.  

```
static const struct counter_driver_api dtmr_cmsdk_apb_api = {
	.start = dtmr_cmsdk_apb_start,
	.stop = dtmr_cmsdk_apb_stop,
	.read = dtmr_cmsdk_apb_read,
	.set_top_value = dtmr_cmsdk_apb_set_top_value,
	.get_pending_int = dtmr_cmsdk_apb_get_pending_int,
	.get_top_value = dtmr_cmsdk_apb_get_top_value,
};
```

andyross, 2024-02-09:
Counter driver enhancement got mislabeled "Timer", bounce to @nordic-krch