zephyrproject-rtos/zephyr #24713: ztest_test_fail() doesn't always work

andrewboie, 2020-04-26:
**Describe the bug**
I started to notice that sometimes in tests a `zassert` expression would fail, and this would be reported to the console output, but the test scenario would still pass.

Upon further inspection, I found that the zassert was being invoked from a child thread,
but the main ztest_thread kept running, and the entire suite passed even though I had
multiple failures.

**Expected behavior**
I expect when ztest_test_pass(), ztest_test_fail(), ztest_test_skip() are invoked, this is not clobbered by some other call to these same APIs.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

showstopper, tests can report passing when they are not

carlescufi, 2020-04-28:
See #24714 for a partial? fix?

andrewboie, 2020-04-28:
@carlescufi is that the right GH link? #24714 is unrelated to this I think (although I encountered this problem while developing it)

andrewboie, 2020-05-06:
I have a partial fix for this in #25011 I think we can lower priority if that goes in.

Better still would be a kernel API to iterate over all nonessential threads and abort them, but there isn't a safe/robust way to do this currently.

pabigot, 2020-05-18:
@carlescufi the partial fix has been merged, should this be lowered in priority as suggested above?

andrewboie, 2020-05-19:
I'll just close this, the few test setups that create complex hierarchies of threads need to be written to clean up after themselves properly and can be dealt with on a case-by-case basis if they don't

closed 2020-05-19 by andrewboie