zephyrproject-rtos/zephyr #24722: OnePlus 7T & peripheral_hr on NRF52 conn failure

alexandru-porosanu-nxp, 2020-04-27:
**Describe the bug**
While attempting to connect a OnePlus 7T (Android 10, Build # 10.0.9.HD65AA) with a  NRF52 (PCA10040) running peripheral_hr on latest head, the connection is established but no heart-rate data is being exchanged. Using the nrf sniffer, one can see that after the connection is established (packet #2382 in the attached capture), there is an exchange of Empty PDUs, but starting at #2405, the Master keeps re-transmitting the same PDU up until the connection is broken at #2617.
I think it's worthwhile to note that this happens only if 2 Mbps PHY is enabled; if the support is disabled, the connection is fine.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52_pca10040
3. west flash

Open "NXP IOT toolbox" (it should be the same behavior for NRF Connect, but I haven't tested), select the Heart Rate monitor and connect to the Nordic board

**Expected behavior**
The heart-rate data should be visible on the phone screen, steadily increasing.

**Impact**
Annoyance

**Screenshots or console output**
```
*** Booting Zephyr OS build zephyr-v2.2.0-1967-g8aeb8a381480  ***
Bluetooth initialized
Advertising successfully started
Connected
[00:00:00.020,000] <inf> bt_hci_core: HW Platform: reserved (0x0000)
[00:00:00.020,000] <inf> bt_hci_core: HW Variant: unknown (0x0000)
[00:00:00.020,000] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 99
[00:00:00.020,000] <wrn> bt_hci_core: No static addresses stored in controller
[00:00:00.170,000] <inf> bt_hci_core: Identity: ca:26:5e:18:cd:57 (random)
[00:00:00.170,000] <inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1
[00:00:00.170,000] <inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff
[00:00:00.690,000] <inf> hrs: HRS notifications enabled
Disconnected (reason 0x08)
Connected
[00:00:20.270,000] <inf> hrs: HRS notifications disabled
[00:00:20.690,000] <inf> hrs: HRS notifications enabled
Connected
Disconnected (reason 0x08)
```
**Environment (please complete the following information):**
 - OS: Linux, FC31
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 8aeb8a381480

[Oneplus7T-Nordic-Peripheral_HR-failure.zip](https://github.com/zephyrproject-rtos/zephyr/files/4538242/Oneplus7T-Nordic-Peripheral_HR-failure.zip)

carlescufi, 2020-04-27:
@alexandru-porosanu-nxp could you please gather a sniffer trace of the connection establishment?
If you have a second board you can use the sniffer here:

https://www.nordicsemi.com/Software-and-Tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE

alexandru-porosanu-nxp, 2020-04-27:
> @alexandru-porosanu-nxp could you please gather a sniffer trace of the connection establishment?
> If you have a second board you can use the sniffer here:
> 
> https://www.nordicsemi.com/Software-and-Tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE

I've attached a capture that has the scan req/scan rsp/conn req exchange plus the phy change, starting at packet 2382. Is there another capture I should do?

alexandru-porosanu-nxp, 2020-04-27:
One more thing: I'm thinking that maybe this is a OnePlus/Android 10 issue, so if you have another Android based phone that also has the 2 Mbps PHY that you can test with, it'd be great to figure this out.
Also, it would be interesting to test on an Iphone as well...but again, I don't have such a device.

joerchan, 2020-04-27:
The sniffer sees the slave at least. But not the master. Looks like an issue with the OnePlus.
From the trace the slave sees the master.

alexandru-porosanu-nxp, 2020-04-27:
I think that actually when switching to 2 Mbps, the master doesn't "hear" the slave anymore. I've taken a screencap of the Wireshark capture on the exact time when things go awry, and I think no 2 Mbps packet is being heard by the master. The blue dotted one is the first packet the master is sending at 2 Mbps. The next one is the slave responding (properly), and then the master re-transmits forever:

[image]

I found something similar described here: https://devzone.nordicsemi.com/f/nordic-q-a/44933/connection-drop-immediately-in-specific-android-models but I don't think it's the same thing as here.

As a side-note, same board with a VEGA board work just fine on 2Mbps.

carlescufi, 2020-04-28:
@alexandru-porosanu-nxp @joerchan can we close this then?

alexandru-porosanu-nxp, 2020-04-28:
I think so, but I'd feel a bit better if somebody can test the 2 Mbps scenario with another device (it looks as it's a OnePlus specific issue)

carlescufi, 2020-05-06:
Tested on master peripheral_hr with an iPhone 11, switch to 2Mbit/s works fine and notifications flow freely after.

closed 2020-05-06 by carlescufi