zephyrproject-rtos/zephyr #24727: Bluetooth mesh bug default controller

johnsmithecon, 2020-04-27:
**Describe the bug**
I’ve got an interrupt on a gpio pin, and when that interrupt gets triggered it publishes a message to the mesh network. When I plug wire into this gpio pin, but have wire connected to nothing (just the air), my interrupt is triggering like crazy. This is normal probably because I am detecting noise. When I let it trigger like crazy for about 20 seconds, I eventually get the following message printed over serial on the device that is receiving the messages over the mesh network: “bt_mesh_net: Out of relay buffers”. 
After I see the out of relay buffers thing, I see “bt_mesh_beacon: Unable to allocate beacon buffer” and then I stop receiving messages. 
I am using default controller if that helps anyone.

**To Reproduce**
Reproduction is described above.

**Expected behavior**

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
Zephyr SDK

**Additional context**
Add any other context about the problem here.

LingaoMeng, 2020-04-27:
@johnsmithecon can put you code to public(github repo), so let’s us find bugs easily.

johnsmithecon, 2020-04-27:
Sorry I’m pretty new to git/github. What does it mean to put code to public? I need to commit to your guys’ repo?

LingaoMeng, 2020-04-28:
> Sorry I’m pretty new to git/github. What does it mean to put code to public? I need to commit then push to your guys’ repo?

see https://github.com/zephyrproject-rtos/zephyr/issues/24311

carlescufi, 2020-04-28:
@johnsmithecon if you can just post the code you are using to trigger this issue somewhere, be it as an attachment, gist or branch anywhere, so we can reproduce locally.

carlescufi, 2020-05-14:
@johnsmithecon closing this for now. Please reopen if this is still an issue.

closed 2020-05-14 by carlescufi