zephyrproject-rtos/zephyr #24731: Bluetooth: controller: Network privacy not respected when address resolution is disabled

joerchan, 2020-04-27:
**Describe the bug**
The resolving list is not used when address resolution has been disabled by the host.
The address resolution enable command should (after a few erratas) only control resolving of RPAs.
If a device has been added to the resolving list with a valid IRK, and address resolution is enabled. This should act as a blacklist, and we should not connect to it, or accept connection requests from it.
Since we cannot resolve it, and using it's identity is a violation of network privacy.

**To Reproduce**
Steps to reproduce the behavior:
Uncomment the following line in `tests/bluetooth/bsim_bt/edtt_ble_test_app/tests_scripts/ll.set1.test_list` and run it according to the readme:
`#LL/CON/INI/BV-24-C #currently failing, to be investigated`

**Expected behavior**
Initiator should not connect to devices in the resolving list with a valid IRK when address resolution is disabled.

**Impact**
Failing LL conformance testcase.

joerchan, 2020-04-27:
The test has a TSE waiver, but that is not related to why this test is failing now. But would explain why we have qualified without this test before:
https://www.bluetooth.org/tse/errata_view.cfm?errata_id=11429

carlescufi, 2021-01-21:
We might need to revisit this the next time we run Controller qualification tests.

PR #32254 (2021-02-14): Bluetooth: controller: Fix network privacy with resolution disabled
When address resolution is disabled, an identity address has
been added into the resolving list with peer IRK, and device
privacy has not been selected for the peer device then
connection indication shall not be sent to the peer that is
advertising using its identity address.

Fixes #24731.
commit fde9ba5c (2021-02-14): Bluetooth: controller: Fix network privacy with resolution disabled

When address resolution is disabled, an identity address has
been added into the resolving list with peer IRK, and device
privacy has not been selected for the peer device then
connection indication shall not be sent to the peer that is
advertising using its identity address.

Fixes #24731.
commit 8ff077a6 (2021-02-19): Bluetooth: controller: Fix network privacy with resolution disabled

When address resolution is disabled, an identity address has
been added into the resolving list with peer IRK, and device
privacy has not been selected for the peer device then
connection indication shall not be sent to the peer that is
advertising using its identity address.

Fixes #24731.
closed 2021-02-14 by nashif