zephyrproject-rtos/zephyr #24733: Misconfigured environment

Nukersson, 2020-04-27:
Building on of the existing examples leads to misconfigured environment error.

Steps to reproduce on mac OS X 10.15.4:

1. Checkout master using tutorial [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html#get-the-code)
( in my case: West version: v0.7.2 and master of Zephyr's project )

2. Run build of an existing example project:
`west build -b nucleo_l432kc samples/hello_world`

3. Get the error:
```
[0/1] Re-running CMake...

Including boilerplate (Zephyr base (cached)): /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.2.99 (/Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr)
-- Board: nucleo_l432kc
-- Found west: /usr/local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found toolchain: gnuarmemb (/usr/local/bin)
CMake Error at /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/cmake/compiler/gcc/generic.cmake:8 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  

User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: gnuarmemb

  Internal variables:

  CROSS_COMPILE: /usr/local/bin/bin/arm-none-eabi-

  TOOLCHAIN_HOME: /usr/local/bin

Call Stack (most recent call first):
  /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/cmake/generic_toolchain.cmake:70 (include)
  /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/cmake/app/boilerplate.cmake:502 (include)
  /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
/usr/local/Cellar/cmake/3.17.1/bin/cmake --regenerate-during-build -S/Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/sampl
es/hello_world -B/Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/build
ninja: error: rebuilding 'build.ninja': subcommand failed
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/bu
ild
```

Obviously the error is in environment variable: 

> CROSS_COMPILE: /usr/local/bin/bin/arm-none-eabi-

Following temporar fix works for me:

In the file: 
>/Users/<your_user_name_here>/Documents/DevProjects/zephyrproject/zephyr/cmake/compiler/gcc/generic.cmake

add following command:

> set(CROSS_COMPILE "/usr/local/bin/arm-none-eabi-")

carlescufi, 2020-04-27:
@Nukersson is your GNU Arm Embedded toolchain installed in `/usr/local/bin` ? We don't support generic installations of GNU Arm Embedded, you need to download it manually from:

https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm

and then install it in a fixed directory and point to it via the `GNUARMEMB_TOOLCHAIN_PATH` environment variable.

Nukersson, 2020-04-27:
> is your GNU Arm Embedded toolchain installed in /usr/local/bin ?
Yes it is.
But what I am confused of is:

> CROSS_COMPILE: /usr/local/bin/bin/arm-none-eabi-
has `bin` twice while `TOOLCHAIN_HOME: /usr/local/bin`. How can it be?

carlescufi, 2020-04-27:
@Nukersson how did you install GNU Arm Embedded?

Nukersson, 2020-04-28:
My GNU Arm Embedded environment installed according to [toolchain-gnuarmemb](https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#toolchain-gnuarmemb)

carlescufi, 2020-04-28:
> P.S: I've updated west/zephyr today and the behavior is no more there.

Thanks, will close the issue then.

closed 2020-04-28 by carlescufi

Alex-EEE, 2021-08-31:
HI I have the same issue.

Alex-EEE, 2021-08-31:
CROSS_COMPILE: /Users/alex/toolchains/arm/bin/bin/arm-none-eabi-

Why is it bin/bin?

Nukersson, 2021-08-31:
>   CROSS_COMPILE: /Users/alex/toolchains/arm/bin/bin/arm-none-eabi-
> 
> Why is it bin/bin?

Oh no, just copy and paste mistake. 😄 I have edited it.

Alex-EEE, 2021-09-01:
Hi, 

 Here are my env vars:
 (dsp-quantized-too-big-fix) benchmark-fw $ echo $GNUARMEMB_TOOLCHAIN_PATH
/Users/alex/toolchains/arm

Is there some way to reset any cached things in the build system?  I did have a different env var at the start.  Here is the full output, showing the vesrion of Zephry

 (dsp-quantized-too-big-fix) benchmark-fw $ ZEPHYR_BASE=~/work/tool-repos/zephyr/zephyr/ west build -p auto -b stm32f4_disco .
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /Users/alex/work/tool-repos/zephyr/zephyr/cmake/app/boilerplate.cmake
CMake Deprecation Warning at /Users/alex/work/tool-repos/zephyr/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
  The OLD behavior for policy CMP0079 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  /Users/alex/work/tool-repos/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /Users/alex/work/tool-repos/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:10 (find_package)

-- Application: /Users/alex/work/edgeimpulse/studio/dsp-pipeline/server/benchmark-fw
-- Zephyr version: 2.6.99 (/Users/alex/work/tool-repos/zephyr/zephyr), build: zephyr-v2.6.0-2865-g74cc8bee7c56
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: stm32f4_disco
-- Cache files will be written to: /Users/alex/Library/Caches/zephyr
-- Found dtc: /usr/local/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/Users/alex/toolchains/arm/bin)
CMake Error at /Users/alex/work/tool-repos/zephyr/zephyr/cmake/compiler/gcc/generic.cmake:8 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: gnuarmemb

  Internal variables:

  CROSS_COMPILE: /Users/alex/toolchains/arm/bin/bin/arm-none-eabi-

  TOOLCHAIN_HOME: /Users/alex/toolchains/arm/bin

Call Stack (most recent call first):
  /Users/alex/work/tool-repos/zephyr/zephyr/cmake/generic_toolchain.cmake:42 (include)
  /Users/alex/work/tool-repos/zephyr/zephyr/cmake/app/boilerplate.cmake:552 (include)
  /Users/alex/work/tool-repos/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /Users/alex/work/tool-repos/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:10 (find_package)

Alex-EEE, 2021-09-01:
Ok, I figured it out.  If you need to change your toolchain path (GNUARMEMB_TOOLCHAIN_PATH) you need to delete your build folder first.  Otherwise, west uses some cached copy of that path

Nukersson, 2021-09-01:
> Ok, I figured it out.  If you need to change your toolchain path (GNUARMEMB_TOOLCHAIN_PATH) you need to delete your build folder first.  Otherwise, west uses some cached copy of that path

It is always a very good idea to remove your build folder after each toolchain / west update and / or git pull.

WayneKeenan, 2022-03-04:
It's not just a good idea, is necessary

bid-p, 2023-08-13:
Ran into issue:

> west build -p always -b adafruit_feather_stm32f405 samples/basic/blinky
> -- west build: making build dir /Users/sidp/zephyrproject/zephyr/build pristine
> -- west build: generating a build system
> Loading Zephyr default modules (Zephyr base).
> -- Application: /Users/sidp/zephyrproject/zephyr/samples/basic/blinky
> -- CMake version: 3.27.2
> -- Found Python3: /Users/sidp/.virtualenvs/.venv/bin/python3.11 (found suitable exact version "3.11.4") found components: Interpreter 
> -- Cache files will be written to: /Users/sidp/Library/Caches/zephyr
> -- Zephyr version: 3.4.99 (/Users/sidp/zephyrproject/zephyr)
> -- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
> -- Board: adafruit_feather_stm32f405
> -- Found toolchain: gnuarmemb (/opt/homebrew/bin/arm-none-eabi-as)
> CMake Error at /Users/sidp/zephyrproject/zephyr/cmake/compiler/gcc/generic.cmake:9 (message):
>   Zephyr was unable to find the toolchain.  Is the environment misconfigured?
> 
>   User-configuration:
> 
>   ZEPHYR_TOOLCHAIN_VARIANT: gnuarmemb
> 
>   Internal variables:
> 
>   CROSS_COMPILE: /opt/homebrew/bin/arm-none-eabi-as/bin/arm-none-eabi-
> 
>   TOOLCHAIN_HOME: /opt/homebrew/bin/arm-none-eabi-as
> 
> Call Stack (most recent call first):
>   /Users/sidp/zephyrproject/zephyr/cmake/modules/FindHostTools.cmake:107 (include)
>   /Users/sidp/zephyrproject/zephyr/cmake/modules/dts.cmake:9 (find_package)
>   /Users/sidp/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
>   /Users/sidp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
>   /Users/sidp/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
>   CMakeLists.txt:4 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake -DWEST_PYTHON=/Users/sidp/.virtualenvs/.venv/bin/python -B/Users/sidp/zephyrproject/zephyr/build -GNinja -DBOARD=adafruit_feather_stm32f405 -S/Users/sidp/zephyrproject/zephyr/samples/basic/blinky

Fixed by setting `GNUARMEMB_TOOLCHAIN_PATH=/opt/homebrew`, as my toolchain was installed through homebrew.