zephyrproject-rtos/zephyr #24744: k_thread_join() taking a very long time on qemu_cortex_m3

cfriedt, 2020-04-27:
**Describe the bug**
`k_thread_join()` is taking a long time to return (corresponding to timeout value for some reason) 

**To Reproduce**
Steps to reproduce the behavior:
1. build and run the attached test case
2. see error

**Expected behavior**
As per the api documentation:
"This API returns immediately if the thread isn’t running"

**Impact**
unnecessary delays

**Screenshots or console output**
```
starting test - test_pipe_and_poll_write_block
3130 T: 0x200008cc test_pipe_and_poll_write_block(): 128: setting up context
3130 T: 0x200008cc test_pipe_and_poll_write_block(): 133: creating secondary thread
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 139: created secondary thread 0x20000054
3140 T: 0x200008cc k_pipe_write_avail(): 33: pipe: 0x200000e4 N: 5 u: 0 r: 0 w: 0
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 168: calling k_pipe_put()
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 173: have written 1 bytes
3140 T: 0x200008cc k_pipe_write_avail(): 33: pipe: 0x200000e4 N: 5 u: 1 r: 0 w: 1
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 168: calling k_pipe_put()
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 173: have written 2 bytes
3140 T: 0x200008cc k_pipe_write_avail(): 33: pipe: 0x200000e4 N: 5 u: 2 r: 0 w: 2
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 168: calling k_pipe_put()
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 173: have written 3 bytes
3140 T: 0x200008cc k_pipe_write_avail(): 33: pipe: 0x200000e4 N: 5 u: 3 r: 0 w: 3
3140 T: 0x200008cc test_pipe_and_poll_write_block(): 168: calling k_pipe_put()
3150 T: 0x200008cc test_pipe_and_poll_write_block(): 173: have written 4 bytes
3150 T: 0x200008cc k_pipe_write_avail(): 33: pipe: 0x200000e4 N: 5 u: 4 r: 0 w: 4
3150 T: 0x200008cc test_pipe_and_poll_write_block(): 168: calling k_pipe_put()
3150 T: 0x200008cc test_pipe_and_poll_write_block(): 173: have written 5 bytes
3150 T: 0x200008cc k_pipe_write_avail(): 33: pipe: 0x200000e4 N: 5 u: 5 r: 0 w: 0
3150 T: 0x200008cc test_pipe_and_poll_write_block(): 146: the next call to k_pipe_put() would block
3150 T: 0x200008cc test_pipe_and_poll_write_block(): 154: calling k_poll()
3150 T: 0x20000054 reader_thread(): 101: secondary thread running
3150 T: 0x20000054 reader_thread(): 105: have written 5 bytes
4160 T: 0x20000054 reader_thread(): 109: calling k_pipe_get()
4160 T: 0x20000054 reader_thread(): 113: calling k_poll_signal_raise()
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 157: returned from k_poll()
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 161: calling k_poll_signal_check()
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 165: received signal / correct result
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 168: calling k_pipe_put()
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 173: have written 6 bytes
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 176: success!
4160 T: 0x200008cc test_pipe_and_poll_write_block(): 178: calling k_thread_join()
4160 T: 0x20000054 reader_thread(): 116: terminating..
9170 T: 0x200008cc test_pipe_and_poll_write_block(): 181: joined!
PASS - test_pipe_and_poll_write_block
```
**Environment (please complete the following information):**
 - OS: Ubuntu Linux (Bionic)
 - Toolchain: cross-compile (gcc-arm-none-eabi package in Ubuntu)
 - Commit SHA or Version used: 75b42d8a37b09cff7dc088515ad15f5f6657a8d0

**Additional context**

cfriedt, 2020-04-27:
[test_pipe_and_poll.c.txt](https://github.com/zephyrproject-rtos/zephyr/files/4541807/test_pipe_and_poll.c.txt)

andrewboie, 2020-05-04:
Confirmed also broken on qemu_x86, should be an arch-neutral issue. I'm investigating...

andrewboie, 2020-05-04:
Hmm, this works flawlessly with a K_FOREVER timeout.
Whatever the root cause ends up being, need to update test cases to exercise this scenario.

PR #24950 (2020-05-05): kernel: fix issue with k_thread_join() timeouts
If k_thread_join() was passed with an actual timeout value,
and not K_FOREVER, the blocking thread was not being properly
woken up when the target thread exits. The timeout itself
was never aborted, causing the joining thread to remain
un-scheduled until the timeout expires.

Amend the k_thread_join() test cases to check that the join
completed before the provided timeout period expired.

Fixes: #24744
commit f1b5d9db (2020-05-05): kernel: fix issue with k_thread_join() timeouts

If k_thread_join() was passed with an actual timeout value,
and not K_FOREVER, the blocking thread was not being properly
woken up when the target thread exits. The timeout itself
was never aborted, causing the joining thread to remain
un-scheduled until the timeout expires.

Amend the k_thread_join() test cases to check that the join
completed before the provided timeout period expired.

Fixes: #24744
closed 2020-05-05 by andrewboie