zephyrproject-rtos/zephyr #24747: tests/lib/heap fails on ARC nsim_sem nsim_em

andrewboie, 2020-04-27:
```
[  2%] Built target parse_syscalls_target
[  3%] Built target kobj_types_h_target
[  6%] Built target syscall_list_h_target
[  9%] Built target driver_validation_h_target
[ 10%] Built target offsets
[ 11%] Built target offsets_h
[ 11%] Built target zephyr_generated_headers
[ 32%] Built target kernel
[ 34%] Built target app
[ 57%] Built target zephyr
[ 58%] Built target linker_script_target
[ 60%] Built target arch__common
[ 74%] Built target arch__arc__core
[ 77%] Built target arch__arc__core__mpu
[ 89%] Built target lib__libc__minimal
[ 91%] Built target lib__posix
[ 93%] Built target subsys__testsuite__ztest
[ 94%] Built target drivers__serial
[ 96%] Built target zephyr_prebuilt
[ 96%] Built target linker_pass_final_script_target
[100%] Built target zephyr_final
Scanning dependencies of target run
Console now belongs to UART, hit CRTL-] to return to simulator.
*** Booting Zephyr OS build zephyr-v2.2.0-2004-g80b47fb440e8  ***
Running test suite lib_heap_test
===================================================================
starting test - test_small_heap
Testing small (2048 byte) heap
successful allocs: 2001/2137 (94%), frees: 1959,  avg usage: 884/2048 (43%)
PASS - test_small_heap
===================================================================
starting test - test_fragmentation
Testing maximally fragmented (2048 byte) heap
successful allocs: 1132/3128 (36%), frees: 968,  avg usage: 980/2048 (48%)
PASS - test_fragmentation
===================================================================
starting test - test_big_heap
Testing big (174762 byte) heap
```

And times out.

vonhust, 2020-05-02:
@andrewboie  i think  the root cause is that nsim is too slow, and 512K SAM is allocated for nsim, . I ran the tests for 10 mins, and got the following result:
```
===================================================================
starting test - test_small_heap
Testing small (2048 byte) heap
successful allocs: 2001/2137 (94%), frees: 1959,  avg usage: 884/2048 (43%)
PASS - test_small_heap
===================================================================
starting test - test_fragmentation
Testing maximally fragmented (2048 byte) heap
successful allocs: 1132/3128 (36%), frees: 968,  avg usage: 980/2048 (48%)
PASS - test_fragmentation
===================================================================
starting test - test_big_heap
Testing big (174762 byte) heap
successful allocs: 3129/3258 (96%), frees: 838,  avg usage: 73657/174762 (42%)
PASS - test_big_heap
===================================================================
Test suite lib_heap_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

If reduce the BIG_HEAP_SIZE,  it will be faster

andrewboie, 2020-05-02:
10 minutes is a long time...@andyross any ideas?

andrewboie, 2020-05-02:
cc @npitre you guys have been working with this code.

andyross, 2020-05-04:
This is definitely a CPU-bound test, it's exhaustively stressing the heap with random allocation patterns.  It completes in 10-20 seconds IIRC on most of the qemu platforms.  Is nsim that much slower or are we hitting some weird performance case?  There was a similar situation with renode, which was much slower than qemu.  There we just disabled that simulator as we had coverage for the architecture in qemu.

FWIW: changing the heap size (which is sized to available memory by default) shouldn't affect the run time of the test except on very small systems.  The iteration count is sized from the "small" heap, which is 2kb max.

vonhust, 2020-05-04:
@andy0808 , my observation is heap size will affect the run time, maybe in sys_heap_alloc.  When I change the MEMSZ from 170 KB (256 * 2/3) to the fixed 16 KB.  It runs faster.

andyross, 2020-05-04:
That's definitely weird.  I'd be tempted to say "cache effects", but obviously this is nsim not hardware.

The iteration count really shouldn't be changing, you can add a printk to verify if you like.  So if it's slower when operating in a bigger heap (which should be **faster**!  bigger heaps have more succeeding allocations, and finding a fitting block is the fast path vs. searching the list and failing) then something else is going on.

Is the larger heap area pushing nsim into a different memory model or something?  Maybe there's a mistake in the linker script and some of the heap is spilling into something nsim thinks is MMIO space which requires special handling?

Is nsim that much slower than qemu's emulated platforms on the smaller test too?

(FWIW: I have no particular reason to demand that you use the 170kb heap, the test runs fine with whatever size, and we have plenty of coverage of large heap spaces on other platforms.  So this is a fine workaround.  But we should figure out what's going on.)

andyross, 2020-05-04:
Oh, I forgot: the test code runs a full `sys_heap_validate()` for every operation, which iterates over the whole heap.  So yes, that's going to be make the run time O(N^2) in the heap size.

Still, qemu_x86 and mps2_an385 (our typical "big" emulated platforms) use a ~256k heap at runtime and complete in ~5 seconds or so.

Is nsim really 100x slower than qemu?  I still think something weirder must be going on.

andrewboie, 2020-05-05:
@wentongwu noticed this test also takes a long time in QEMU if we enable icount.

andyross, 2020-05-05:
Maybe what we need is a framework for identifying cpu-heavy tests and cycle-exact/slow emulators and minimizing the overlap?  I mean, for this one there's minimal need to run this on every board.  It's a unit test of pure C code; one platform per architecture is fine

npitre, 2020-05-06:
On Tue, 5 May 2020, Andy Ross wrote:

> Maybe what we need is a framework for identifying cpu-heavy tests and cycle-exact/slow emulators and minimizing the overlap?

Is the notion of wall clock time accurate enough on those emulators? If 
so there could be a reasonable maximum time for the test to run, and if 
timeout is reached without errors then this is OK.

andyross, 2020-05-09:
> Is the notion of wall clock time accurate enough on those emulators?

I'm guessing not.  With qemu -icount, by default, the emulator is decoupled from real world time and runs it's simulated time as fast as the CPU allows.  There is a mode where it will stuff delays in its CPU idle instructions to try to align with realtime, but that only works when it's able to keep up at all, which isn't the case here.

That's certainly true on nsim too, since it seems to be taking 10 minute to emulate a few seconds of hardware time.

What we'd want is an indication of "real world time" from within the emulator, which (even if it's available) would require writing a whole extra driver to give it to us.

npitre, 2020-05-11:
In that case maybe the test harness could be set to consider a so 
qualified test as successful if it reaches its timeout without flagging 
any errors. And such tests could have a much smaller but still 
reasonable timeout.

carlescufi, 2020-05-21:
@vonhust @andyross any updates on this one?

andyross, 2020-05-21:
My recommendation remains whitelisting this test on cycle-exact simulators.  It completes in a few seconds on hardware or qemu's default code generator, but looks to be two orders of magnitude slower with icount/renode/nsim.

And this is fundamentally a library test.  It doesn't interact with hardware or drivers at all, not even to do things like check time.  All we care is that we have coverage for the behavior on at least one platform per hardware architecture, and we have (non-icount) qemu for arc.

vonhust, 2020-05-25:
According to above discussion, I label it as a low prio bug

PR #26691 (2020-07-07): tests: filter out nsim as it's very slow to run lib_heap test
nsim is slow, if the MEMSZ is too targe, the test will
run a long time and make sanitycheck timeout.

There are two possible fixes to pass the sanitycheck test
  * filter out nsim, not to block the sanitcheck test
  * use a small MEMSZ for nsim to reduce the execution time.

Considering there are potential improvements for nsim (because
some qemu targets can pass), we use the 1st fix to pass the
sanitycheck temporarily.

Fixes #24747
closed 2020-07-07 by carlescufi