zephyrproject-rtos/zephyr #24750: device_list_get should not include the initialization failed devices

wentongwu, 2020-04-28:
**Describe the bug**
device_list_get located in device sub system is the API exposed to other modules, and the returned list shouldn't include the initialization failed devices, though it's not common case that device fail it's initialization.

pabigot, 2020-04-28:
This API just provides the start and end of the linker section that contains all devices.  It doesn't synthesize a new list of any sort.

It should be possible to iterate over the devices in that list and determine whether they've been initialized successfully or not.  This may tie to #19448 which also needs to know device status.

carlescufi, 2020-04-28:
@wentongwu consensus is that we should document this properly so that it explicitly states that this returns the full list of devices, started or not, and then it's the responsibility of the caller to walk the array and determine which ones are initialized or not.

wentongwu, 2020-04-30:
> It should be possible to iterate over the devices in that list and determine whether they've been initialized successfully or not, though that capability may not be present. 

yes, we should have that API or there is status flag in struct device.

wentongwu, 2020-04-30:
> @wentongwu consensus is that we should document this properly so that it explicitly states that this returns the full list of devices, started or not, and then it's the responsibility of the caller to walk the array and determine which ones are initialized or not.

Thanks, the title of this issue maybe confuse, my intention is to get the list of succeed initialization device in ether way(device module provide that API or caller iterate over the devices by the status which currently doesn't exist in struct device), currently PM module use device_list_get to suspend/resume device and it maybe not well considered.

pabigot, 2020-04-30:
@wentongwu Agreed, we need #23407 merged then I'll work this, adding a function to check the state and clarifying the documentation.  #19448 needs to be taken into account in the API, though the implementation can wait.

tbursztyka, 2020-04-30:
Just a quick notice, as your discussion caught my attention: driver_api pointer is not going to be used anymore as a marker for initialization success/failure. Reason being that it prevents device driver instance to be "const-ified" to stay in ROM. (I am actually working on this const issue)

> yes, we should have that API or there is status flag in struct device.

Yes a status flag per-device driver instance will be the solution (it will be used for other things as well, maybe we'll be able to pack other things there, it's 32 bits: there is space).

> adding a function to check the state and clarifying the documentation.

I already made a draft RFC on device status reporting, it's still in a very early stage, but the grounds for such check is there already (the function is called device_check_status(), it's not yet part of device public API but your use case would definitely need it).

There is at least this one #24511 (it introduces the status, but only uses it for synchronization purposes), then the status reporting RFC extends its usage to initializtion status, hw fault, etc... Note that stuff in this PR could be used to reduce the PM structure size as well later on, most notably the lock attribute.

PR #26342 (2020-06-23): device: add API to iterate over devices and check their validity
The power management infrastructure added a generic API to access the static device list, but it is not generally available.  Provide an internal standard API for this.

Replace all the shell infrastructure that iterated over devices with a helper that implements it once.  This incidently fixes bugs related to using a pointer before checking it for null, and matching a prefix wherever it occurred in the device name.

Finally add standard API to determine whether a device is OK to use, so we can replace the nulling of the API poin [... cut]
closed 2020-06-23 by carlescufi