zephyrproject-rtos/zephyr #24751: What is purpose of the CONFIG_ADC_X

susundberg, 2020-04-28:
**Describe the bug**
This probably issue in documentation / my understanding but i feel confused what is CONFIG_ADC_1 for example trying to do.

I am testing with stm32_min_dev_blue and the ADC seems to work just fine without enabling the CONFIG_ADC_1. 

While grepping for CONFIG_ADC_xyz from boards/st/ directory it seems some variants have pin-muxing, for example `./nucleo_f103rb/pinmux.c`:
```
#ifdef CONFIG_ADC_1
        {STM32_PIN_PA0, STM32F1_PINMUX_FUNC_PA0_ADC123_IN0},
#endif /* CONFIG_ADC_1 */
```

That apparantly pinmuxes channel 0 (to what is default-on-boot: analog). Then what about other channels? In my tests, for example PB0 that is ADC1_IN8 works just fine.

**Expected behavior**
ADC_1 does not work if the ADC_1 is not enabled.

**Impact**
No impact.

ABOSTM, 2020-04-28:
Hi @susundberg,
You are right, CONFIG_ADC_1 is used to configure pinmux ADC channel to Analog,
but is is also used in drivers/adc/adc_shell.c.
Because Analog is the default mode after reset for this chip, your ADC works. even if you didn't activate CONFIG_ADC_1 and even if you didn't declare your PIN for channel8. So you are lucky.
But you may had trouble using adc shell. 

> Then what about other channels?

We declared only 1 channel as example and you can customize it depending on your needs.
But it would be a shame to declare all 16 channels when only 1 is needed, in some way it would reserve 16 pins.

susundberg, 2020-04-29:
I undestand this - and for my behalf one can close this issue.

But if its to enable ADC1 channel 0 should it not be called ADC1_CH0 ? And on the other hand, why should not there be ADC1_CH8 then? That would also document (i see the pinmux as documentation, of what pin is used but again, this is just me) the specific pin used in the pinmux.c.

(btw: checking another older project with stm32f4 seems to put the pins ins nopull, analog)

Ty for reply.

erwango, 2020-05-04:
@susundberg We're closing this PR as CONFIG_ADC_X symbols are being removed and replaced by dts nodes.

> But if its to enable ADC1 channel 0 should it not be called ADC1_CH0 ? I got confused by the name of ADC1 -- that surely to me seems like peripheral ADC1. And on the other hand, why should not there be ADC1_CH8 then? That would also document (i see the pinmux as documentation, of what pin is used but again, this is just me) the specific pin used in the pinmux.c.

We don't have this granularity.
But I agree current solution is not a prefect fit. Next step is to move pinmux.c to device tree. I hope this will help to make things clearer.

closed 2020-05-04 by erwango