zephyrproject-rtos/zephyr #24803: ADC driver test hangs on atsame54_xpro

stephanosio, 2020-04-29:
**Describe the bug**
The `tests/drivers/adc/adc_api` test hangs on `atsame54_xpro`.

**To Reproduce**
Build and run `tests/drivers/adc/adc_api` on `atsame54_xpro` board.

**Expected behavior**
Test completes successfully.

**Screenshots or console output**
```
Running test suite adc_basic_test
===================================================================
starting test - test_adc_sample_one_channel
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: b7626df483847be61d60e97758b4afff91fb976e

stephanosio, 2020-04-29:
cc @benpicco @nandojve

carlescufi, 2020-05-05:
@stephanosio is this something you plan to take a look at?

stephanosio, 2020-05-06:
> @stephanosio is this something you plan to take a look at?

@carlescufi I plan to eventually get back to this, possibly before 2.3 release.

stephanosio, 2020-08-19:
This issue is still present. This happens because the SAM D/E 5x ADC has two interrupt lines, and the SAM0 ADC driver only connects the first interrupt line.

I have a fix for this in a private repo, and I will eventually upstream it along with the other bug fixes in the future.

closed 2020-11-03 by github-actions[bot]