zephyrproject-rtos/zephyr #24805: On x86, misalligned SSE accesses can occur when multithreading is enabled

wobak-j, 2020-04-29:
**Describe the bug**
The [_dummy_thread_ variable](https://github.com/zephyrproject-rtos/zephyr/blob/d650f4b494c62c1e62cc727f75b2047ed692f636/kernel/init.c#L470) in `z_cstart` is not always aligned to 16 bytes, e.g. when `CONFIG_DEBUG_OPTIMIZATIONS=y` is set. 
This leads to exceptions when performing [_fxsave_](https://github.com/zephyrproject-rtos/zephyr/blob/d650f4b494c62c1e62cc727f75b2047ed692f636/arch/x86/core/ia32/swap.S#L158) while swapping it out in `swap.S`.

**To Reproduce**
1. to _./samples/userspace/shared_mem/proj.conf_, add following configurations:
`CONFIG_DEBUG_OPTIMIZATIONS=y`
`CONFIG_FPU=y`
`CONFIG_SSE=y`
2. mkdir build; cd build
3. cmake -DBOARD=qemu_x86 ./samples/userspace/shared_mem ./build
4. cmake --build build --target all
5. cmake --build build --target run
6. protection fault occurs during swap

**Expected behavior**
Swapping out the dummy thread should not produce a protection fault regardless of the configuration.

**Impact**
This issues makes it impossible to use the debug configuration for userspace applications when newlib is enabled, since it requires SSE.

```
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 11
E: General protection fault (code 0x0)
E: EAX: 0x00125da0, EBX: 0x00009500, ECX: 0x00125b83, EDX: 0x0014b56c
E: ESI: 0x00000000, EDI: 0x00126020, EBP: 0x00000000, ESP: 0x0014b534
E: EFLAGS: 0x00000006 CS: 0x0008 CR3: 0x00125b00
E: call trace:
E: EIP: 0x00100348
E: NULL base ptr
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x0014b56c (unknown)
E: Halting system
```

**Environment:**
 - OS: Ubuntu 19.10
 - Zephyr SDK 0.11.2
 - Zephyr 2.2.0 at least until 4008960afc05e1ca25973ebc585a9d63c10b9f14

**Additional context**
This might be an issue in _x86_64-zephyr-elf-gcc_ in the SDK not aligning the struct to 16 byte on the stack. Since the contained [_tFpRegSetEx_](https://github.com/zephyrproject-rtos/zephyr/blob/af3defd34e1c06e9fdc57da6c1b5c91a95c44023/include/arch/x86/ia32/thread.h#L115) struct has the correct alignment attribute, my understanding is that this should not happen.
Making the _dummy_thread_ variable static produces correct alignment.

andrewboie, 2020-04-29:
Related issue: https://github.com/zephyrproject-rtos/zephyr/issues/17893

carlescufi, 2020-05-05:
@nashif and @andrewboie could this be closed as a duplicate of #17893?

andrewboie, 2020-05-05:
@carlescufi no we should leave this open, this is a statically defined object and it should be possible to set it up with the proper alignment, I'm surprised it isn't already.

andrewboie, 2020-05-05:
We have a dummy_thread created on the interrupt stack in z_cstart(). It appears that at runtime, this is misaligned, for example with this crash:

```
E: General protection fault (code 0x0)
E: EAX: 0x00125da0, EBX: 0x0014b56c, ECX: 0x00125b83, EDX: 0x0014b56c
E: ESI: 0x00000000, EDI: 0x00126020, EBP: 0x00000000, ESP: 0x0014b534
E: EFLAGS: 0x00000006 CS: 0x0008 CR3: 0x00125b00
E: call trace:
E: EIP: 0x00100348
E: NULL base ptr
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x0014b56c (unknown)
E: Halting system
```

In this case, the dummy thread object is at `0x14b55c` which is not 16-byte aligned. The specified 16-byte alignment for thread->arch on x86 is being ignored here.

Two options to fix this:
1. Move dummy_thread to a static toplevel declaration, which we know will be aligned properly. However this memory can never be reclaimed.
2. Move x86's fxsave memory into a reserved region of the stack object. I like this better because it also fixes #17893 at the same time.

I'm going to pursue the second option first.

carlescufi, 2020-05-12:
@andrewboie thanks for the preliminary investigation.
Could we then close #17893 as a duplicate of this one? That one is a low priority bug. If you decide not to close it, should this one be low too?

andrewboie, 2020-05-12:
@carlescufi this is more serious, I'm working on it now for 2.3

andrewboie, 2020-05-12:
Option 2 isn't working out. I posted some details in #17893.
Since this is just x86 I am looking at option 3 now.

PR #25266 (2020-05-13): x86: fix dummy thread alignment
- Consolidate some code to initialize dummy threads into an inline function

- Implement a custom swap to main for x86 32-bit, which still uses a dummy thread but ensures that the memory for it is properly aligned.

- Remove comments which are not correct

Fixes: #24805

Please see individual commit messages for details.
closed 2020-05-13 by carlescufi