zephyrproject-rtos/zephyr #24840: OpenThread is broken

retfie, 2020-04-29:
**Describe the bug**
Devices (nrf52840dk_nrf52840) upgraded to 5321c42882 origin/master unable to connect to OpenThread network anymore. Most probable root cause is this change 19c8ff9f0ac99c957505bbdd32868dcbb25d1944 at least according to git bisect suggestion. Checkout on a hash before it, recovers ability of nodes to connect to OT network.

**To Reproduce**
Steps to reproduce the behavior:
west build -s samples/net/lwm2m_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf ot_sed.conf"

**Expected behavior**
Devices should able to connect to OT network and communicate with other nodes.

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr version: 2.2.99
 - Toolchain gnuarmemb gcc-arm-none-eabi-7-2018-q2-update
 - Commit: 5321c42882 (origin/master ...)

**Additional context**
content of ot_sed.conf

CONFIG_OPENTHREAD_MTD=y
CONFIG_OPENTHREAD_MTD_SED=y
CONFIG_OPENTHREAD_POLL_PERIOD=1000

rlubos, 2020-05-05:
@retfie I think we need more information here, I've just tested echo_client sample, configured as SED (on nrf52840dk_nrf52840) and it attached to the network with no issues at all. I've used current master, as for today (https://github.com/zephyrproject-rtos/zephyr/commit/b78c9dca52c506af3446c0dd3025ac0c9f6dcea2). Do you observe this behavior only with SED devices, or does it happen for FTD (routers) as well?

Several ideas that come to my mind, based on the past experience:
1. When you switch between FTD/MTD devices on the same board, it's good to erase the settings area entirely, otherwise some obsolete settings may lead to unexpected behavior (or even crash). `nrfjprog -e` from the command line, or `ot factoryreset` from Zephyr shell should do the job.
2. MTD devices cannot have statically assigned mesh-local addresses (in Zephyr assigned with `CONFIG_NET_CONFIG_MY_IPV6_ADDR`), as this violates the Thread specification and breaks the attachment procedure (MTDs need to register their addresses at their parent). Probably you hit this?

carlescufi, 2020-05-05:
@retfie 

>Toolchain gnuarmemb gcc-arm-none-eabi-7-2018-q2-update

This is quite an old GCC release. It's likely not the root cause, but upgrading wouldn't hurt. Since this is an issue with an external project that we don't track toolchain requirements for.

retfie, 2020-05-06:
@rlubos here is some test that I've made from my side:

`git fetch origin`
`git checkout origin/master`
`west update -f always`

* c6edfa0971 (HEAD, origin/master, origin/HEAD)

**1. Usecase: echo_client (FTD)**
`rm -rf build/`
`west build -s samples/net/sockets/echo_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"`
`nrfjprog -e`
`west flash -d build --hex-file build/zephyr/zephyr.hex`
ot factoryreset
ot thread stop
ot channel `<ch>`
ot panid `<panid>`
ot networkname `<name>`
ot masterkey `<key>`
ot thread start

got role 2 (child), and node can be ping from border router

**2. Usecase echo_client (MED/SED)**
`rm -rf build/`
`west build -s samples/net/sockets/echo_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf ot_sed_serial.conf"`
`nrfjprog -e`
`west flash -d build --hex-file build/zephyr/zephyr.hex`
ot factoryreset
ot thread stop
ot channel `<ch>`
ot panid `<panid>`
ot networkname `<name>`
ot masterkey `<key>`
ot thread start

got role 2 (child), and node can be ping from border router

`cat samples/net/sockets/echo_client/ot_sed_serial.conf`
CONFIG_OPENTHREAD_MTD=y
CONFIG_OPENTHREAD_MTD_SED=y
CONFIG_OPENTHREAD_POLL_PERIOD=1000
CONFIG_OPENTHREAD_SHELL=y
CONFIG_NET_SHELL=y

**3. Usecase lwm2m-client (FTD)**
`rm -rf build/`
`cp samples/net/sockets/echo_client/overlay-ot.conf samples/net/sockets/echo_client`
`west build -s samples/net/lwm2m_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"`
`nrfjprog -e`
`west flash -d build --hex-file build/zephyr/zephyr.hex`
ot factoryreset
ot thread stop
ot channel `<ch>`
ot panid `<panid>`
ot networkname `<name>`
ot masterkey `<key>`
ot thread start

got role 4 (leader), and node can't be ping from border router

rlubos, 2020-05-06:
@retfie 
So in terms of `lwm2m_client`, you've made no changes at all in the default config? I'm asking, because I've tried to reproduce the issue locally (just the attachment procedure, I don't have a running BR setup right now), and first thing the sample did was a crash.

I copied the file the `overlay-ot.conf` just as you did, and built lwm2m_client "as is" with that overlay. The issue turned out to be the main thread stack size (the sample crashed during initialization due to stack overflow), so I've added the following to `prj.conf` in the `lwm2m_client`:
`CONFIG_MAIN_STACK_SIZE=2048`.

After that, my `lwm2m_client` joined the OT network successfully. Looking at the `prj.conf` of the lwm2m_client, I'd also consider increasing the net buf count, since it's rather small (only 10).

retfie, 2020-05-06:
@rlubos 
In matter of fact I saw such crashes once or two times after re-base my project on top of master, but since there was a lot of my changes and in addition other changes coming from moving my project to  new DT macros API from upstream, I start thinking that I'd screwed it somewhere. Then I moved to original lwm2m sample (with original prj.conf), where I don't have crashes, most probably because I do not make a system reset , just `ot factoryreset` and then looked if device was able to join OT network or not.

Adding more stack solves crash issue after reset and even device able to join to OT network again, so we could consider this issue as solved.
Thanks !

rlubos, 2020-05-07:
> Adding more stack solves crash issue after reset and even device able to join to OT network again, so we could consider this issue as solved.

I'll take the liberty to close it then. Thanks!

closed 2020-05-07 by rlubos