zephyrproject-rtos/zephyr #24842: Support Building on Aarch64

cfriedt, 2020-04-29:
**Is your enhancement proposal related to a problem? Please describe.**
Building on Aarch64 is pretty easy.

**Describe the solution you'd like**
For the build to not try to pass `-m32` to the compiler, unconditionally.

**Describe alternatives you've considered**
Submitting a patch.

**Additional context**
```
2/126] Building C object zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj 
ccache /usr/bin/gcc -DBUILD_VERSION=zephyr-v2.2.0-1657-g0f37c1e23a9c -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I../kernel/include -I../arch/posix/include -I../include -Izephyr/include/generated -I../soc/posix/inf_clock -I../boards/posix/native_posix -I../subsys/net/lib/sockets/. -I../subsys/testsuite/include -I../subsys/testsuite/ztest/include -Os -imacros/home/chrisfriedt/workspace/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacros/home/chrisfriedt/workspace/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/chrisfriedt/workspace/zephyrproject/zephyr/tests/net/socket/socketpair=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/chrisfriedt/workspace/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/chrisfriedt/workspace/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -m32 -include /home/chrisfriedt/workspace/zephyrproject/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -std=c11 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj   -c /home/chrisfriedt/workspace/zephyrproject/zephyr/arch/posix/core/offsets/offsets.c
gcc: error: unrecognized command line option ‘-m32’
ninja: build stopped: subcommand failed.
```

cfriedt, 2020-04-30:
On a side note, it's also pretty trivial to run
```bash
apt-get update
apt install -y gcc-arm-none-eabi
```
and with that, use
```
export ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
export CROSS_COMPILE=/usr/bin/arm-none-eabi-
```

To cross compile from Aarch64 to Aarch32

aescolar, 2020-04-30:
Realted to #13633

aescolar, 2020-04-30:
> On a side note, it's also pretty trivial to run[..]

Would you like to add a paragraph in the documentation (together with your PR) describing how to do it?

cfriedt, 2020-04-30:
> > On a side note, it's also pretty trivial to run[..]
> 
> Would you like to add a paragraph in the documentation (together with your PR) describing how to do it?

No problem. Hopefully the syntax is correct ;-)

PR #24843 (2020-05-09): posix arch: support building on aarch64 / allow host-specific cmake includes
This change enables specific compiler and linker options to be used in the case that an `arch/posix/os.arch.cmake` file exists.

Note: os and arch in the above case are evaluations of [CMAKE_HOST_SYSTEM_NAME](https://cmake.org/cmake/help/latest/variable/CMAKE_HOST_SYSTEM_NAME.html) and [CMAKE_HOST_SYSTEM_PROCESSOR](https://cmake.org/cmake/help/latest/variable/CMAKE_HOST_SYSTEM_PROCESSOR.html).

Otherwise, the existing "generic" compiler and linker flags in `arch/posix/CMakeLists.txt` are used.

Additional flags and ch [... cut]
commit 2c0eecaa (2020-05-09): posix arch: build on aarch64 / allow host-specific cmake includes

This change enables specific compiler and linker options to be used in
the case that an arch/posix/os.arch.cmake file exists.

Note: os and arch in the above case are evaluations of
CMAKE_HOST_SYSTEM_NAME and CMAKE_HOST_SYSTEM_PROCESSOR.

Otherwise, the existing "generic" compiler and linker flags in
arch/posix/CMakeLists.txt are used.

Additional flags and checks are provided in
arch/posix/Linux.aarch64.cmake.

Added scripts/user_wordsize.py to detect if userspace is 64-bit or
32-bit, which should be consistent with the value  [... cut]
closed 2020-05-09 by carlescufi