zephyrproject-rtos/zephyr #24844: Setting esp-idf path to match Espressif's documentation

beriberikix, 2020-04-30:
I set up a new mac with both Espressif's IDF/toolchain and Zephyr. I ran into a build error because the `ESP_IDF_PATH` was not set. This wasn't set correctly because:

1. The Zephyr docs didn't mention it when [setting up the environment](https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html#set-up-build-environment)
2. The Espressif [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#id5) tell you to run an [`install.sh`](https://github.com/espressif/esp-idf/blob/master/install.sh) that sets a different value to `IDF_PATH`.

I'd like to recommend switching the `IDF_PATH` to match the Espressif docs. I think `west` options needs to be updated and some of the CMAKE files in Zephyr.

Thoughts?

carlescufi, 2020-04-30:
@beriberikix would you send a PR to fix the documentation?

carlescufi, 2020-04-30:
@nashif have you ever had to set `ESP_IDF_PATH` when building for ESP32?

beriberikix, 2020-04-30:
Happy to submit the immediate doc bug. However, the change I suggested would require modifying the source code. Should I submit the PR for the docs or should we discuss the proposed change first?

pabigot, 2020-05-03:
I do the following to build for ESP32:

    . ~/esp/esp-idf/export.sh
    export ZEPHYR_TOOLCHAIN_VARIANT=espressif
    export ESPRESSIF_TOOLCHAIN_PATH=${HOME}/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/
    export ESP_IDF_PATH=${IDF_PATH}

    west build -b esp32
    west flash --esp-device /dev/ttyUSB0

closed 2020-07-17 by github-actions[bot]