zephyrproject-rtos/zephyr #24853: os: Precise data bus error with updatehub

mtahirbutt, 2020-04-30:
**Describe the bug**
A clear and concise description of what the bug is.

When I run the updatehub app without DNS and without DHCP, it generates error right after probing

[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: * BUS FAULT *[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: Precise data bus error[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: BFAR Address: 0x6f2e065[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: r0/a1: 0x06f2e065 r1/a2: 0x08039309 r2/a3: 0x00000031[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: r3/a4: 0x06f2e065 r12/ip: 0x20002c0e r14/lr: 0x08027e21[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: xpsr: 0x61000000[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: Faulting instruction address (r15/pc): 0x0802073e[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: Current thread: 0x20004f04 (unknown)[1B][0m
[1B][8D[1B][J[00:04:33.023,000] [1B][1;31m<err> os: Halting system[1B][0m

What have you tried to diagnose or workaround this issue?

I removed the the recent enhancements (probe security, cleaning the code warnings etc.) made to updatehub lib and the app works fine at the moment. It is related to memory leak. So I tried to play around memory allocations in the updatehub lib and finally the code runs well.
Please see my branch 
https://github.com/mtahirbutt/zephyr/tree/disco_l475_iot1

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=disco_l475_iot1
3. make
4. See error

**Expected behavior**

The updatehub client should successfully download the image for updatehub-CE

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

Right now, I am successful in downloading the image and the framework works fine with older version of updatehub lib with some changings.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: * BUS FAULT *[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: Precise data bus error[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: BFAR Address: 0x6f2e065[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: r0/a1: 0x06f2e065 r1/a2: 0x08039309 r2/a3: 0x00000031[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: r3/a4: 0x06f2e065 r12/ip: 0x20002c0e r14/lr: 0x08027e21[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: xpsr: 0x61000000[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: Faulting instruction address (r15/pc): 0x0802073e[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[1B][0m
[1B][8D[1B][J[00:04:32.948,000] [1B][1;31m<err> os: Current thread: 0x20004f04 (unknown)[1B][0m
[1B][8D[1B][J[00:04:33.023,000] [1B][1;31m<err> os: Halting system[1B][0m

**Environment (please complete the following information):**
 - OS: (e.g. Ubuntu 18.04.4 LTS)
 - Toolchain (e.g Zephyr 2.2.99 SDK-0.11.2, ...)
 - Commit SHA: 3223c43af240e21a5cc6ec64c3afb97f7c7157e3 

**Additional context**
I did not use DHCP and DNS.

nandojve, 2020-04-30:
CC @otavio

carlescufi, 2020-05-05:
@nandojve or @otavio any chance you can take a look at the branch and logs?
@mtahirbutt if you haven't tried that yet, please enable `CONFIG_HW_STACK_PROTECTION` and retry, in case it's a stack overflow.

mtahirbutt, 2020-05-05:
I also reduced k_malloc (inside probe and install functions) assignments in the updatehub lib code, the issue also vanishes. will update but still facing some more issues. i can also try CONFIG_HW_STACK_PROTECTION by the way.

nandojve, 2020-05-05:
Hi @carlescufi, we are looking at. I already self assigned myself the issue.
Thank you for reach us.

nandojve, 2020-05-08:
@mtahirbutt We need that #25010 fixed to continue testing.

mtahirbutt, 2020-05-08:
@nandojve  I am experiencing some other error for zephyr master also "SHA256 size not valid etc." during Probe metadata also "could not parse jason etc." to the most recent zephyr so stopped at that and now working with my branch I mentioned. I have to make that work to another network interface. will look into zephyr master later.

otavio, 2020-05-08:
@mtahirbutt we apologize for the delay but we are on a dead-end as this is not reproducible here, so we foresee two possible scenarios:

1) we are missing something
2) your code/settings are different and it causes the side effect

On top of that, we found multiple issues with the ST MCUs and we've been reporting and fixing those issues as we can. Last is #25010 as @nandojve mentioned.

So, the bottom line is, we need to be using very same code/settings and reproduce the issue to be capable of debugging it.

mtahirbutt, 2020-05-09:
@otavio thanks for your note. Yes there are still some side effects (retransmissions and in appropriate memory usage) with my branch that I will take care later. right now I want to updatehub work quickly with another interface, will see zephyr master (about probe security and other enhancements made by you) later.

PR #25603 (2020-05-26): lib: updatehub: Fix bus fault
Fix variable-size string copy patch that introduced a runtime bug that
causes a bus fault.

Fixes #24853.

CC @otavio @mtahirbutt
commit ba88c996 (2020-05-26): lib: updatehub: Fix bus fault

Fix variable-size string copy patch that introduced a runtime bug that
causes a bus fault.

Fixes #24853.
closed 2020-05-26 by carlescufi