zephyrproject-rtos/zephyr #24859: os: Add memory partition overlap assert check is not made for x86 boards

maksimmasalski, 2020-04-30:
**Describe the bug**
When using in code memory domain API k_mem_domain_add_partition(), in its implementation function during check of the memory partition assertions, I found out that assertion which checks that memory partitions are not overlapped is not executing for x86 boards.
1. In _kernel/mem_domain.c_ to check that it is true, I added debug printk before line L160
```
#if defined(CONFIG_EXECUTE_XOR_WRITE) || \                  
        defined(CONFIG_MPU_REQUIRES_NON_OVERLAPPING_REGIONS)
        printk("\nOVERLAP ASSERT CHECKED\n");               
        __ASSERT(sane_partition_domain(domain, part), "");  
#endif                                                      
```
2. Also in _kernel/mem_domain.c_ I added one more debug printk before line L50
```
#if defined(CONFIG_MPU_REQUIRES_NON_OVERLAPPING_REGIONS)                        
                /* Partitions overlap */                                        
                printk("\nDEEP OVERLAP ASSERT CHECK\n");                        
                __ASSERT(false, "overlapping partitions <%lx...%x>, <%lx...%x>",
                        part->start, last,                                      
                        parts[i].start, cur_last);                              
                return false;                                                   
#endif                                                                          
```
Then I found out that printk message was not printed when code is running on x86. I understand, that for x86 that Assertion should be checked too. When I'm trying to set CONFIG_MPU_REQUIRES_NON_OVERLAPPING_REGIONS=y in prf.conf file for x86 architecture, during the build I receive error.

**To Reproduce**
Steps to reproduce the behavior:
1. Add printk in parts of code _kernel/mem_domain.c_ shown above
2. Run 
$west build -p -b reel_board tests/kernel/mem_protect/mem_protect_domains/
and then run 
$west build -p -b qemu_x86 tests/kernel/mem_protect/mem_protect_domains/
3. Check the difference in output of printk for reel_board and qemu_x86

**Expected behavior**
Expect that overlap assert check must be executed for all platforms.

**Impact**
x86 is not checked for memory partition overlap

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 19.10
 - Zephyr SDK 0.11.2
 - Commit d1f51db366507956afa6dde6206fedd1a86b04b6

carlescufi, 2020-05-05:
This is not a bug, `CONFIG_MPU_REQUIRES_NON_OVERLAPPING_REGIONS` is not enabled for x86 because of the architecture itself. If you want to add additional checks for x86 that are outside of the conditionally-compiled code please send a PR o log an enhancement issue.

andrewboie, 2020-05-05:
> This is not a bug, CONFIG_MPU_REQUIRES_NON_OVERLAPPING_REGIONS is not enabled for x86 because of the architecture itself. If you want to add additional checks for x86 that are outside of the conditionally-compiled code please send a PR o log an enhancement issue.

It's out of spec, the spec for memory domains indicates none of the partitions overlap. There is no defined precedence for overlapping regions and you can get an undefined configuration even if the HW nominally supports it.

maksimmasalski, 2020-07-06:
@andrewboie Did you implement any changes for the memory partitions to close that issue?

andrewboie, 2020-08-27:
This is now fixed in master branch , an assertion should be failed regardless of the target arch.

closed 2020-08-27 by andrewboie