zephyrproject-rtos/zephyr #24868: Enhancement of ARM CMSDK Dual Timer driver

kkiyota, 2020-04-30:
**Is your enhancement proposal related to a problem? Please describe.**
Existing ARM CMSDK Dual Timer driver, zephyr/drivers/counter/timer_dtmr_cmsdk_apb.c, has limited capabilities such as mode setting(currently only periodic and wrap mode), registers' access, only timer1 available(can't access to timer2), etc.

**Describe the solution you'd like**
Add additional APIs in zephyr/drivers/counter/timer_dtmr_cmsdk_apb.c or add flexibility to access registers.

**Describe alternatives you've considered**
Allow local customization. 

**Additional context**
Current APIs
```
static const struct counter_driver_api dtmr_cmsdk_apb_api = {
	.start = dtmr_cmsdk_apb_start,
	.stop = dtmr_cmsdk_apb_stop,
	.read = dtmr_cmsdk_apb_read,
	.set_top_value = dtmr_cmsdk_apb_set_top_value,
	.get_pending_int = dtmr_cmsdk_apb_get_pending_int,
	.get_top_value = dtmr_cmsdk_apb_get_top_value,
};
```

kkiyota, 2020-05-01:
In addition, I also would like to ask to enhance it to support multiple instances.  My SoC has two dual timers in it.

galak, 2020-05-01:
@kkiyota there isn't much active development on the CMSDK drivers at this time.  We're happy to take any PR to address the enhancements you are asking for.