zephyrproject-rtos/zephyr #24869: qemu_x86: with icount enabled, crash in test_syscall_torture

andrewboie, 2020-04-30:
**Describe the bug**
After applying the following patch:

```
diff --git a/boards/x86/qemu_x86/board.cmake b/boards/x86/qemu_x86/board.cmake
index ba98880ef9..108f88251a 100644
--- a/boards/x86/qemu_x86/board.cmake
+++ b/boards/x86/qemu_x86/board.cmake
@@ -19,7 +19,8 @@ set(QEMU_FLAGS_${ARCH}
   -device isa-debug-exit,iobase=0xf4,iosize=0x04
   ${REBOOT_FLAG}
   -nographic
-  )
+  -icount shift=7,align=off,sleep=off -rtc clock=vm
+)
 
 # TODO: Support debug
 # board_set_debugger_ifnset(qemu)
```

I noted a consistent failure on the test which hammers the CPU with system calls:

```
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 11
*** Booting Zephyr OS build zephyr-v2.2.0-2142-g8b27d5c6b906  ***
Running test suite syscalls
===================================================================
starting test - test_string_nlen
PASS - test_string_nlen
===================================================================
starting test - test_string_nlen
E: syscall z_vrfy_string_nlen failed check: Memory region 0x00122280 (size 24) read access denied
PASS - test_string_nlen
===================================================================
starting test - test_to_copy
E: syscall user_copy failed check: Memory region 0x00122260 (size 32) write access denied
PASS - test_to_copy
===================================================================
starting test - test_user_string_copy
E: string too long 0x0010f010 (32)
E: syscall user_copy failed check: Memory region 0x00122280 (size 24) read access denied
PASS - test_user_string_copy
===================================================================
starting test - test_user_string_alloc_copy
E: string too long 0x0010f010 (32)
E: syscall z_user_alloc_from_copy failed check: Memory region 0x00122280 (size 24) read access denied
PASS - test_user_string_alloc_copy
===================================================================
starting test - test_arg64
PASS - test_arg64
===================================================================
starting test - test_syscall_torture
Running syscall torture test with 4 threads on 1 cpu(s)
E: Page fault at address 0x0 (error code 0x10)
E: Linear address not present in page tables
E:   PDE: 0x0000000000132827 Writable, User, Execute Enabled
E:   PTE: Non-present
E: EAX: 0xfffffff5, EBX: 0x0000700f, ECX: 0x001225e0, EDX: 0x00122094
E: ESI: 0x000000cf, EDI: 0x0015dfd8, EBP: 0x00137398, ESP: 0x0015dfe0
E: EFLAGS: 0x00003a16 CS: 0x0008 CR3: 0x001222c0
E: call trace:
E: EIP: 0x00000000
E:      corrupted? (bp=0x00137398)
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x00122128 (unknown)
E: Halting system
```

I don't know if this is reproducible without the icount patch, but my feeling this is a kernel bug.

andrewboie, 2020-04-30:
@wentongwu FYI

andrewboie, 2020-04-30:
EIP is 0. This is being incorrectly reported as a stack overflow. ESP points to a location 0x0015dfe0 on the trampoline stack, so I'm suspecting a KPTI issue:

```
0015d120 D z_trampoline_stack_start
0015e000 D __device_init_start
0015e000 D __device_PRE_KERNEL_1_start
0015e000 d __device_sys_init_spec_ctrl_init0
0015e000 D z_shared_kernel_page_end
0015e000 D z_trampoline_stack_end
```

andrewboie, 2020-04-30:
> This is being incorrectly reported as a stack overflow. 

To fix this, z_x86_check_stack_bounds() needs to also check if the provided addr falls within the trampoline stack for the current CPU.

andrewboie, 2020-04-30:
Trampoline stack contents:

```
0x15dfe0:	0x00103a34	0x0015dfe4	0x00000000	0x00100836
0x15dff0:	0x0000002b	0x00000246	0x00130394	0x00000033
```

So we have a stack frame of:

```
    SS 0x00000033
   ESP 0x00130394	
EFLAGS 0x00000246	
    CS 0x0000002b
   EIP 0x00100836 <- instruction after the 'int 0x80' in to_copy syscall invocation
       0x00000000
       0x0015dfe4 <- ??? this is past _image_ram_end. Maybe heap location?
       0x00103a34 <- text, intstub.S line 110
```

The first five values match the sort of exception stack frame we would get from invoking the custom "to_copy" syscall defined by the test case. Which makes sense, the test hammers the CPU with syscalls from several threads, and to_copy() is one of them, the test defines several custom syscalls.

The last two values look deeply suspicious to me, like we got an irq or something at a bad time.

carlescufi, 2020-05-01:
@andrewboie would you mind labelling test-related issues that affect a particular test or set of tests with the "Tests" label? This will help triage for the release

closed 2020-05-01 by andrewboie

andrewboie, 2020-05-04:
> @andrewboie would you mind labelling test-related issues that affect a particular test or set of tests with the "Tests" label? This will help triage for the release

@carlescufi I'd rather not....shouldn't we only mark issues/PRs as "tests" if the tests were somehow part of the issue? In this case, the test was not the problem.

carlescufi, 2020-05-26:
> @carlescufi I'd rather not....shouldn't we only mark issues/PRs as "tests" if the tests were somehow part of the issue? In this case, the test was not the problem.

Well, no, as per the agreement and email I sent, "Tests" means "a test fails". We can of course change that, see my email to the mailing list to continue the thread there.

PR #24879 (2020-05-01): x86: fix trampoline stack clobber
We need to lock interrupts before setting the thread's
stack pointer to the trampoline stack. Otherwise, we
could unexpectedly take an interrupt on this stack
instead of the thread stack as intended.

The specific problem happens at the end of the interrupt,
when we switch back to the thread stack and call swap.
Doing this on a per-cpu trampoline stack instead of the
thread stack causes data corruption.

Fixes: #24869
commit dac61f45 (2020-05-01): x86: fix trampoline stack clobber

We need to lock interrupts before setting the thread's
stack pointer to the trampoline stack. Otherwise, we
could unexpectedly take an interrupt on this stack
instead of the thread stack as intended.

The specific problem happens at the end of the interrupt,
when we switch back to the thread stack and call swap.
Doing this on a per-cpu trampoline stack instead of the
thread stack causes data corruption.

Fixes: #24869