zephyrproject-rtos/zephyr #24889: stm32f469i discovery board and samples/display/lvgl fails

tristenator, 2020-05-01:
hello, i'm new to embedded development and i don't know if this is a bug, but when i build and flash   samples/display/lvgl for the stm32f469i discovery board the console result is:

*** Booting Zephyr OS build zephyr-v2.2.0-478-gcd679452b20e  ***
[00:00:00.005,000] <err> lvgl: Display device not found.
[00:00:00.005,000] <err> app: device not found.  Aborting test.

i used:

west build -b stm32f469i_disco samples/display/lvgl
west flash

and i tried a different sample:

west build -b stm32f469i_disco samples/drivers/display

but got a lot of compilation errors:

`
/home/tristen/zephyrproject/zephyr/samples/drivers/display/src/main.c:183:35: error: 'DISPLAY_DEV_NAME' undeclared (first use in this function)
  183 |  LOG_INF("Display sample for %s", DISPLAY_DEV_NAME);
      |                                   ^~~~~~~~~~~~~~~~
/home/tristen/zephyrproject/zephyr/include/logging/log_core.h:230:6: note: in definition of macro 'Z_LOG_TO_PRINTK'
  230 |    ##__VA_ARGS__);          \
      |      ^~~~~~~~~~~
/home/tristen/zephyrproject/zephyr/include/logging/log_core.h:287:2: note: in expansion of macro '__LOG'
  287 |  __LOG(_level,           \
      |  ^~~~~
/home/tristen/zephyrproject/zephyr/include/logging/log.h:61:24: note: in expansion of macro 'Z_LOG'
   61 | #define LOG_INF(...)   Z_LOG(LOG_LEVEL_INF, __VA_ARGS__)
      |                        ^~~~~
/home/tristen/zephyrproject/zephyr/samples/drivers/display/src/main.c:183:2: note: in expansion of macro 'LOG_INF'
  183 |  LOG_INF("Display sample for %s", DISPLAY_DEV_NAME);
      |  ^~~~~~~
/home/tristen/zephyrproject/zephyr/samples/drivers/display/src/main.c:183:35: note: each undeclared identifier is reported only once for each function it appears in
  183 |  LOG_INF("Display sample for %s", DISPLAY_DEV_NAME);
      |                                   ^~~~~~~~~~~~~~~~
/home/tristen/zephyrproject/zephyr/include/logging/log_core.h:230:6: note: in definition of macro 'Z_LOG_TO_PRINTK'
  230 |    ##__VA_ARGS__);          \
      |      ^~~~~~~~~~~
/home/tristen/zephyrproject/zephyr/include/logging/log_core.h:287:2: note: in expansion of macro '__LOG'
  287 |  __LOG(_level,           \
      |  ^~~~~
/home/tristen/zephyrproject/zephyr/include/logging/log.h:61:24: note: in expansion of macro 'Z_LOG'
   61 | #define LOG_INF(...)   Z_LOG(LOG_LEVEL_INF, __VA_ARGS__)
      |                        ^~~~~
/home/tristen/zephyrproject/zephyr/samples/drivers/display/src/main.c:183:2: note: in expansion of macro 'LOG_INF'
  183 |  LOG_INF("Display sample for %s", `DISPLAY_DEV_NAME);`

this is the first time i've tried using the display on the board, but i can confirm for other samples and basic non-display programs the board is operational, eg:

west build -b stm32f469i_disco samples/hello_world
west flash

the console result is:

*** Booting Zephyr OS build zephyr-v2.2.0-478-gcd679452b20e  ***
Hello World! stm32f469i_disco
Hello World! stm32f469i_disco
Hello World! stm32f469i_disco
Hello World! stm32f469i_disco

i'm on linux (fedora core 31) and any help/advice would be much appreciated, thank you.

Laczen, 2020-05-02:
Hi @tristenator, the board you are using has no definition of a display. The dts file should be modified to include the display definition. As an example how the display should be defined you can look at the reel_board dts file (for the reel board the display is connected to spi).

tristenator, 2020-05-03:
ok thank you @Laczen i'll take a look, appreciate the help

closed 2020-05-03 by tristenator