zephyrproject-rtos/zephyr #24908: Shared object files or dynamic libraries

rfcard, 2020-05-02:
I would like to be able to create dynamic libraries (".so" shared object files) that my Zephyr application can download when it needs functionality.

Is it possible to do this today, but in another way?

MaikVermeulen, 2021-03-31:
I know this is old, but I'm very interested in this.

carlescufi, 2021-04-30:
We would need a bit more information as to what you would like to do.
If you want examples on how to distribute code as shared libraries (`.a`) you can take a look at the [sdk-nrfxlib](https://github.com/nrfconnect/sdk-nrfxlib) repository of the Nordic nRF Connect SDK, which is pulled by the main [sdk-nrf manifest](https://github.com/nrfconnect/sdk-nrf/blob/master/west.yml)

carlescufi, 2021-04-30:
Note that Zephyr does not support dynamic linking at all, so the `.a` files need to be present at build time