zephyrproject-rtos/zephyr #24910: kernel: stack sentinel crashes

sslupsky, 2020-05-03:
I noticed a peculiar issue while doing some debugging.  I have the stack sentinel turned on.  I am debugging on a Arduino samd21 board (Cortex M0+).  I noticed that the stack sentinel always causes a hard fault.  The result of this prevents the kernel message indicating the fault from ever reaching the console.  Is there a "safe" way to use the stack sentinel?

This screen shot illustrates the stack sentinel exception:
[image]

LOG_PANIC() at the top of the call stack above triggers the following sequence of events that results in the kernel fault:
[image]

andrewboie, 2020-05-03:
This isn't really the information we need.
What we need is the actual error messages that were logged when the fault happened. Preferably in text and not screenshot form.
If it is an MPU fault, I can tell you right now that you cannot use MPU stack guards, and the stack sentinel at the same time (nor would you want to since they do the same thing, with the MPU being far more robust)

andrewboie, 2020-05-03:
If you can't capture any console data, can you check if CONFIG_MPU_STACK_GUARD is also enabled?

You could also enable CONFIG_LOG_MINIMAL which directs all log messages to printk() synchronously instead of buffering them.

sslupsky, 2020-05-04:
@andrewboie I will try to capture some more information about this.  The problem I am encountering is there are no console messages when the faults happen.  I recall only one time observing a console message indicating a hard fault or stack fault or any other kind of fatal fault.  I have seen a fault message occasionally on the J-Link RTT viewer when it is connected during a debug session but the USB_CDC connection that I regularly use for my shell and console doesn't receive the panic messages so I never see them.  The screen shots above show that when the logging system goes into panic mode, it appears to encounter another fault.

Does CONFIG_MPU_STACK_GUARD  work with Cortex-M0+?  I think this code in fault.c excludes M0+?
```
#if defined(CONFIG_ARMV6_M_ARMV8_M_BASELINE)
/* HardFault is used for all fault conditions on ARMv6-M. */
#elif defined(CONFIG_ARMV7_M_ARMV8_M_MAINLINE)

#if defined(CONFIG_MPU_STACK_GUARD) || defined(CONFIG_USERSPACE)
u32_t z_check_thread_stack_fail(const u32_t fault_addr,
	const u32_t psp);
#endif /* CONFIG_MPU_STACK_GUARD || defined(CONFIG_USERSPACE) */
```

sslupsky, 2020-05-04:
CONFIG_MPU_STACK_GUARD is not enabled.

andrewboie, 2020-05-04:
I can't help you further without the complete text of the fatal error message.

You should not rule out the possibility that this is all working as intended, and the fatal error is a stack overflow detected by the sentinel mechanism.

Did you enable CONFIG_LOG_MINIMAL ?

sslupsky, 2020-05-04:
I do not have CONFIG_LOG_MINIMAL turned on at the moment.  I will see if that helps. 

I think you are correct, in the use case I outline above the stack sentinel did indeed detect a stack overflow.  What I think may be a problem is that I do not see the corresponding console output indicating that a stack overflow occurred.  Is my understanding correct?  Does the kernel send a fatal message to the console when it detects a stack overflow that indicates the stack overflow condition?

andrewboie, 2020-05-05:
> Does the kernel send a fatal message to the console when it detects a stack overflow that indicates the stack overflow condition?

Yes, it triggers the kernel fatal error path.

carlescufi, 2020-05-21:
@sslupsky is this resolved and, if not, can you add the info required?

sslupsky, 2020-05-30:
@carlescufi The problem continues to be an issue.  I have not had the time yet to get back to this and investigate further why this is happening.  My current understanding is that when the stack sentinel fires, it puts the log system into panic mode.  When that happens, the stack depth appears to go crazy which leads to another fault. I am not sure there is some recursion going on or if this is an expected amount of depth when operating in panic mode.  Either way, it seems that it makes the stack sentinel difficult to use on sam0.

andrewboie, 2020-05-31:
> Does CONFIG_MPU_STACK_GUARD work with Cortex-M0+?

As far as I know the MPU on Cortex-M0 is enabled.

Have you checked if CONFIG_MPU_STACK_GUARD is enabled in your project?
Look at `zephyr/.config` in your build directory.

sslupsky, 2020-06-01:
I believe that is what I checked and it was not enabled.  I will have another look but I do not think M0 has MPU.

I encountered this problem again over the weekend.  I think the underlying issue is that the shell loops endlessly when there is a log panic.  Seems related to the USB CDC and polling mode.  I also run the Segger backend for my logging so have two backends running if I fire up the debugger.  But unfortunately, the shell seems to lock things up.

andrewboie, 2020-06-02:
> I believe that is what I checked and it was not enabled. I will have another look but I do not think M0 has MPU.

Oh that's right, you did check earlier. Mea culpa.

Lots of people use stack sentinel without issues (even though at heart it's a dirty hack) I wonder if the trouble you are having is due to something with the specifics of how log backends or other debug stuff is set up in your configuration.

One thing to try on your board is to run the test `tests/kernel/fatal` on your board using `sentinel.conf` in the test directory and see whether it passes.

nzmichaelh, 2020-06-12:
Hi @sslupsky the SAM0 USB driver has a null pointer bug and a buffer overflow bug.  This might be related.  Try patching in #26020 and #25879 and see if that fixes the problem.

andrewboie, 2020-06-24:
I'm closing this. Please re-open if you are able to provide an example with clear steps to reproduce. There's not enough information yet for someone else to do root-cause analysis.

closed 2020-06-24 by andrewboie