zephyrproject-rtos/zephyr #24911: arch: arm: aarch32: When CPU_HAS_FPU for Cortex-R5 is selected, prep_c.c uses undefined symbols

Sanji1981, 2020-05-03:
**Describe the bug**
When compiling for a Cortex-R5-based SoC which selects CPU_HAS_FPU option, the z_arm_floating_point_init() function defined in /arch/arm/aarch32/prep_c.c makes use of the SCB (defined in a CMSIS header for Cortex-M3) which is undefined when CPU_CORTEX_R is selected.

**Expected behavior**
Builds fails with:
```
2: error: ‘SCB’ undeclared (first use in this function)
  SCB->CPACR &= (~(CPACR_CP10_Msk | CPACR_CP11_Msk));
```

**Impact**
Advance in the development of current project.

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 18.04(
 - Toolchain (Linaro GCC 7.13)
 - Tag zephyr-2.2.0, [top commit](https://github.com/zephyrproject-rtos/zephyr/commit/9518bd19b7a736bd649fa2fc8d8d52967c0b486f)

stephanosio, 2020-05-03:
Enabling FPU (`CONFIG_FPU=y`) is not supported at the moment. This is WIP. See https://github.com/zephyrproject-rtos/zephyr/issues/20594 and https://github.com/zephyrproject-rtos/zephyr/issues/19979.

stephanosio, 2020-05-03:
> **Impact**
> Advance in the development of current project.

I will probably upstream the Cortex-R FPU support next week.

Sanji1981, 2020-05-04:
@stephanosio  Thank you very much for the prompt reply and please forgive me for the bothering question. That will be very helpful, I'd like to thank you in advance as well.

closed 2020-05-04 by Sanji1981