zephyrproject-rtos/zephyr #24916: echo_client sample return: Cannot connect to TCP remote (IPv6): 60 (frdm_k64f <--> native_posix)

lmajewski, 2020-05-03:
**Describe the bug**
I'm trying to setup echo_server and echo_client for **native_posix** (on my host PC) and **frdm_k64f** development board.
What have you tried to diagnose or workaround this issue?
To fix the problem - I've followed steps described in: https://github.com/zephyrproject-rtos/zephyr/issues/24282

**To Reproduce**
Setup native_posix:
https://docs.zephyrproject.org/latest/guides/networking/native_posix_setup.html#networking-with-native-posix
Create zeth2.conf in net-tools:
```
# Configuration file for setting IP addresses for a network interface.

INTERFACE="$1"

HWADDR="00:00:5e:00:53:ff"

IPV6_ADDR_1="2001:db8::4"
IPV6_ROUTE_1="2001:db8::/64"

IPV4_ADDR_1="192.0.2.4/24"
IPV4_ROUTE_1="192.0.2.0/24"

ip link set dev $INTERFACE up

ip link set dev $INTERFACE address $HWADDR

ip -6 address add $IPV6_ADDR_1 dev $INTERFACE nodad
ip -6 route add $IPV6_ROUTE_1 dev $INTERFACE
ip address add $IPV4_ADDR_1 dev $INTERFACE
ip route add $IPV4_ROUTE_1 dev $INTERFACE > /dev/null 2>&1
```
```sudo ./net-setup.sh --config zeth2.conf```

Create bridge:
```brctl addbr zeth-br && brctl addif zeth-br zeth && brctl addif zeth-br enxe46f13 && ifconfig zeth-br up```

The IP setup:
- frdm_k64f: 192.0.2.2
- zeth interface 192.0.2.4
- native_posix 192.0.2.1

Build native_posix:
```west build --pristine -b native_posix zephyr/samples/net/sockets/echo_server -- -DCONFIG_NET_CONFIG_MY_IPV4_GW=\"192.0.2.4\"```
```west build -t run```

Frdm
 ```west build --pristine -b frdm_k64f zephyr/samples/net/sockets/echo_client/ -- -DCONFIG_NET_CONFIG_MY_IPV4_GW=\"192.0.2.4\"```

Ping between native_posix and frdm_k64f works correctly - I've checked it with wireshark.
When I try to test client / server communication I do receive following error:
```<err> net_echo_client_sample: Cannot connect to TCP remote (IPv6): 60```

**Expected behavior**
echo_client / echo_server works correctly to test the network development on my frdm_k64f board.

**Impact**
Testing of my network changes is a bit tricky now - I would prefer to use Zephyr standard test/
sample code.

**Environment (please complete the following information):**
 - OS:  Linux,
 - Toolchain (Zephyr SDK - 0.11.2)
 - Commit SHA or Version used - zephyr SHA1:  e737761d23c00172b76af4821b63edfd0bbc0df3

jukkar, 2020-05-05:
> When I try to test client / server communication I do receive following error:
> `<err> net_echo_client_sample: Cannot connect to TCP remote (IPv6): 60`

Note that you might need to change the IP addresses in samples prj.conf files as you have changed the network IP addressing from default.

jukkar, 2020-05-05:
Last time I investigated this, the issue was that both zephyr instances were using same MAC address which prevented the connection. But the pings work in your setup, then that is not the issue then.
Other issue that I have seen, is that Linux host has prevented packet flow, so I had to adjust firewall rules.

lmajewski, 2020-05-06:
I do not have any firewall in my setup. It is the USB dongle 10/100 ETH, simple L2 switch and my board. Pings are working for this setup (as the gateway is setup properly). It seems like there is some other issue.

jukkar, 2020-05-09:
Created PR #25136 that adds information how to connect QEMU / native_posix boards together. Hopefully it helps for this issue.

dleach02, 2020-08-17:
@lmajewski, did @jukkar PR change your results here?

lmajewski, 2020-08-24:
@dleach02 In the end I've decided to do the example for DSA with using LLDP, so I did not delved into this example at all.

closed 2020-08-24 by lmajewski