zephyrproject-rtos/zephyr #24918: Segger RTT using j-link doesn't work on NXP i.MX RT

arvid-r, 2020-05-03:
**Describe the bug**
Debug output, and logging using Segger RTT and j-link hardware doesn't work on i.MX RT SOCs from NXP. More info about the RTT subsystem here: [https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_RTT_CONSOLE.html](RTT)

**To Reproduce**
Steps to reproduce the behavior:
1. Select any sample application that runs on iMX RT 1020 EVK board.
2. Add the following to prj.conf:
`CONFIG_UART_RTT=y
CONFIG_UART_CONSOLE=n
CONFIG_USE_SEGGER_RTT=y`
3. Build and flash the sample application.
4. Start the j-link logger using the following command:
`JLinkRTTLogger -device MIMXRT1021xxx5A -if SWD -speed 4000 -rttchannel 0 /dev/stdout`

**Expected behavior**
JLinkRTTLogger automatically finds the ring buffers required for its operation in the device's memory, and the Zephyr startup output can be seen in the terminal.

**Actual behavior**
JLinkRTTLogger doesn't find the ring buffer and no output can be seen from the device.

**Impact**
Debug printouts using RTT is impossible.

**Environment:**
- Any OS

**Cause**
The linker will place the ring buffer in external SRAM, but JLinkRTTLogger will only scan internal RAM segments (as it seems) for the buffers.

**Workaround**
The map file can be scanned for a symbol called `_SEGGER_RTT` and the exact adress can be fed 
to JLinkRTTLogger executable. A better solution would be to to use `CODE_DATA_RELOCATION` to force this ring buffer into internal memory

carlescufi, 2020-05-05:
@arvid-r is this something you'd consider sending a PR for?

arvid-r, 2020-05-05:
Yes! I have a fix. I just need to figure out some details. Can I submit a PR targeting the Segger repo for example? And then, how do I make sure that the CODE_DATA_RELOCATION dependency is handled. I can't find any Kconfig files for the Segger repo.

carlescufi, 2020-05-06:
You can send a PR against https://github.com/zephyrproject-rtos/segger @arvid-r

closed 2020-07-22 by github-actions[bot]