zephyrproject-rtos/zephyr #24926: Remove all uses of CONFIG_LEGACY_TIMEOUT_API  from the tree before 2.3

carlescufi, 2020-05-04:
Subsystems to do:

#24861:
- [ ] BBC micro:bit display
- [ ] Video
- [ ] UART async

TBD (@alexanderwachter):
- [ ] CAN
- [ ] ISOTP

Mani-Sadhasivam, 2020-05-04:
@carlescufi For LoRa, conversion is done in #24649

carlescufi, 2020-05-04:
> @carlescufi For LoRa, conversion is done in #24649

Thank you! Updated

carlescufi, 2020-05-04:
@andrewboie could you port the CMSIS RTOS API please?
@pfalcon could you do `lib/posix`?

pfalcon, 2020-05-05:
> @pfalcon could you do lib/posix please or let me know if you can't?

I will look into it. I assume the timeframe is "before the release"?

I also wish the description pointed to a "model" commit doing such a refactor.

carlescufi, 2020-05-05:
> > @pfalcon could you do lib/posix please or let me know if you can't?
> 
> I will look into it. I assume the timeframe is "before the release"?

Thanks! Yes, there is still time during the 2.3 stabilization period

> I also wish the description pointed to a "model" commit doing such a refactor.

Right, sorry. https://github.com/zephyrproject-rtos/zephyr/pull/24861 could be a starting point, though I see you have already pushed a PR! Thank you!

pfalcon, 2020-05-06:
@carlescufi 

>  though I see you have already pushed a PR

Indeed, https://github.com/zephyrproject-rtos/zephyr/pull/24964 was posted, but that appears to systematically fail CI due to https://github.com/zephyrproject-rtos/zephyr/issues/25017. I would need help with driving resolution for that issue.

carlescufi, 2020-05-06:
Thanks @andyross for #24956. Marked as done now.

closed 2020-05-08 by carlescufi