zephyrproject-rtos/zephyr #24928: Running Zephyr Bot tests on local machine

Nukersson, 2020-05-04:
It would be very helpful to have possibility for typical Zephyr Bot tests ( e.g. CI-Test) running on local development machine. Thus each developer can save many time especially at the beginning.

Probably a Docker image would be suitable for this purpose.

If docker approach is not possible, some CI-Test run before pull request on forked branch may be helpful.

Does anybody have some other ideas or already existing solutions in mind?

P.S.: I am currently a beginner in Zephyr World. Thus my Feature Request may be just a question to already existing solution.

Best Regards,
Alex

carlescufi, 2020-05-04:
@Nukersson you can already do that today very easily with the `sanitycheck` script that is provided with Zephyr.

Nukersson, 2020-05-04:
@carlescufi Do you mean this [ sanity check docs](https://docs.zephyrproject.org/latest/guides/test/sanitycheck.html)?

carlescufi, 2020-05-04:
> @carlescufi Do you mean this [ sanity check docs](https://docs.zephyrproject.org/latest/guides/test/sanitycheck.html)?

Yes, exactly.

carlescufi, 2020-05-04:
@Nukersson you can also pull down the docker image that is used in CI.

>docker run -ti -v $HOME/Work/github/zephyr:/workdir \
docker.io/zephyrprojectrtos/zephyr-build:latest

carlescufi, 2020-05-04:
Closing this since it is already addressed.

closed 2020-05-04 by carlescufi