zephyrproject-rtos/zephyr #24939: LSPI of NXP i.MX RT timing and interrupt issue

lgl88911, 2020-05-04:
**Describe the bug**
1. Zephyr Can't update DBT/SCKPCS/PCSSCK to nxp_hal according to spi rate， cause spi transfer slow.
2. The delay time calculated by nxp_hal has an overflow problem，will casue no update  DBT/SCKPCS/PCSSCK,cause spi transfer slow.
3. Other interrupts are treated as transfer completion interrupts ,Causes spi_write to return immediately

I submitted 2 pr to solve these problems：
https://github.com/zephyrproject-rtos/hal_nxp/pull/36
https://github.com/zephyrproject-rtos/zephyr/pull/23918

carlescufi, 2020-05-21:
@MaureenHelm could you please take a look?

MaureenHelm, 2020-05-22:
@lgl88911 will you please split this into two separate GitHub issues:
1. Configure/optimize the delay time. This is an enhancement. The overflow can be addressed as part of the enhancement, since it's not a problem otherwise.
2. Other interrupts treated as transfer completion. This is a bug. Can you elaborate on how to reproduce this, and the priority?

lgl88911, 2020-05-23:
@MaureenHelm  Thanks for your reply
1. for delay time:When the spi speed is configured to be relatively high, the delay calculated by 500K will greatly reduce the transmission rate. The following is an example of 5M. Looking at the overall transmission at a time, the actual transmission speed is only slightly higher than 500K, but far below 5M .
[image]
 2. After the master transfer is completed,if the sr flag is not clear, the callback of the transfer
completion will be called again in other interrupts. You can use the following method to reproduce：
       Call spi_wirte about 150K Byte , Measure the time used by spi_write, under normal circumstances it will take some time, this time will become 0 once issue occur. The occurrence of issues is random and cannot be recovered once it occur. The speed of SPI is set between 20M ~ 25M, which is relatively easy to copy.

lgl88911, 2020-05-23:
@MaureenHelm 
I raise https://github.com/zephyrproject-rtos/zephyr/issues/25566 to tracking interrupt issue. This issue to keep tracking time delay issue

PR #26504 (2020-07-02): drivers: spi: Add optional delays to mcux dspi and lpspi drivers
Adds optional device tree properties to set delays between spi chip
select assert/deassert and clock edges in the mcux dspi and lpspi
drivers. If these properties are not set, then the minimum supported
delays are used.

Verified that tests/drivers/spi/spi_loopback/ still passes on
mimxrt1050_evk (lpspi driver) and frdm_k64f (dspi driver).

~~Still need to verify delay timing with a scope.~~

Fixes #24939
Alternative to #23918

cc: @lgl88911
closed 2020-07-02 by nashif