zephyrproject-rtos/zephyr #24960: The example "blinky" didn't work on MIMXRT1050-EVK

jasonedn, 2020-05-05:
**Describe the bug**
The led didn't blink 

**To Reproduce**
Build the project as below:
west build -p auto -b mimxrt1050_evk samples/basic/blinky

**Expected behavior**
The build passed but the led didn't blink after downloading. The codes didn't crash and ran properly

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS:  Linux and Windows
 - Zephyr SDK
 - Commit SHA: f3b4d8a

**Additional context**
The user led should link when the codes run

MaureenHelm, 2020-05-05:
I'm not able to reproduce this on the latest master (f89defea51ad1ed895cf8306ae55aeebf92ada56) or on f3b4d8a86f7810249486bfe17a3b372db2580b86. Can you check that SW7 is configured to boot from Hyperflash?

From [MIMXRT1050-EVK User Guide](https://www.nxp.com/webapp/Download?colCode=IMXRT1050EVKBHUG):

[image]

jasonedn, 2020-05-06:
I think the boot mode setting is correct and the blinky example from NXP
MCUXpresso worked. First, the imixrt1050_evk default configuration doesn't
enable the LED as below because the node enet is enabled. This doesn't
cause this issue because enet happens to initialize the led pin. But it
should be a bug for button example.  Second, the gpio output registers
DR_SET and DR_CLEAR used in gpio_mcux_gpio.c do NOT work on my
imixrt1050_evk RevA. I also verified it with MCUXpresso. I don't know why
it worked on your board. Only when the output register DR is used, it
worked.
static int mimxrt1050_evk_init(struct device *dev)
{
    ARG_UNUSED(dev);

    CLOCK_EnableClock(kCLOCK_Iomuxc);
    CLOCK_EnableClock(kCLOCK_IomuxcSnvs);

#if !DT_HAS_NODE(DT_NODELABEL(enet))
    /* LED */
    IOMUXC_SetPinMux(IOMUXC_GPIO_AD_B0_09_GPIO1_IO09, 0);

    IOMUXC_SetPinConfig(IOMUXC_GPIO_AD_B0_09_GPIO1_IO09,
                IOMUXC_SW_PAD_CTL_PAD_PKE_MASK |
                IOMUXC_SW_PAD_CTL_PAD_SPEED(2) |
                IOMUXC_SW_PAD_CTL_PAD_DSE(6));

    /* SW0 */
    IOMUXC_SetPinMux(IOMUXC_SNVS_WAKEUP_GPIO5_IO00, 0);

MaureenHelm, 2020-06-25:
> the gpio output registers DR_SET and DR_CLEAR used in gpio_mcux_gpio.c do NOT work on my imixrt1050_evk RevA. I also verified it with MCUXpresso.

@jhqian are there any errata related to this?

jasonedn, 2020-06-26:
This is a silicon issue. Rev A doesn't support it but Rev B does.

dleach02, 2020-07-10:
Can we close this as "not supported" rev of the silicon?

jasonedn, 2020-07-12:
Yes, I think so

closed 2020-07-12 by jasonedn