zephyrproject-rtos/zephyr #24961: i2c_write allows non-i2c structs to be passed without any errors

jeremyherbert, 2020-05-05:
**Describe the bug**
I just chased down a bug where the `i2c_write` function was passed a device binding which was not an I2C device. There was no indication that this was an error (no compiler warning/errors, no runtime assertion failures) but there really should be something to prevent this.

**To Reproduce**
```
struct device *dev;
uint8_t dummy[1] = {0};

dev = device_get_binding(LED_PORT); // this is a GPIO, not I2C
gpio_pin_configure(dev, LED, GPIO_DIR_OUT);

i2c_write(dev, dummy, 1, 0x23); // this runs fine, and even returns success (== 0)!
```
	
**Expected behavior**
Some sort of error or assertion failure should occur

**Impact**
- Can cause bugs which are difficult to track down as it fails unsafe (it returns success despite being an incorrect driver)
- Makes spelling errors in variable names easy to miss

**Environment**
 - OS: Ubuntu 18.04
 - Toolchain: gnuarmemb
 - Commit: ff214d637e
 - Board: nrf52840_pca10056

stephanosio, 2020-05-05:
This issue is not specific to `i2c_write`.

Zephyr does not use device-specific type, so obviously no static type checking is done.

As for dynamic type checking, since there is no "device type" value in `struct device`, so there is currently no straight-forward method for API functions to validate the specified device instance type (which, I believe, is a deliberate decision made to reduce memory footprint and checking overhead).

If we are really concerned about this, we could potentially add a build option to enable dynamic device type validation (cc @pabigot @carlescufi).

This sounds more like an enhancement request than a bug.

jeremyherbert, 2020-05-05:
I see your point, but having i2c_write return the success error code when it didn’t actually succeed seems like a bug to me.

stephanosio, 2020-05-05:
> I see your point, but having i2c_write return the success error code when it didn’t actually succeed seems like a bug to me.

The result of providing a non-I2C device instance to the I2C API functions, when there is no safeguard against it, is "unpredictable", and "returning success" just happens to be one of such unpredictable cases.

That is not a bug, but rather an inherent (and deliberate) limitation of the current Zephyr device API implementation.

jeremyherbert, 2020-05-05:
Yes, I think you are right then. 

Was the plan to have some sort of polymorphism for drivers, or is it purely to save RAM?

pabigot, 2020-05-05:
A system call from usermode would have validated that the device passed was an I2C device, but in privileged mode (which is the default) no such validation is performed.

tbursztyka, 2020-05-05:
This could be added but as a debug feature only.

vChavezB, 2022-11-28:
Just an extra thought about this ,  an assert would be nice to detect at compile time. Metadata could be added about the type of node (uart-controller, gpio-controller,spi-controller,etc) type and only used as mentioned only in debug mode `#ifndef NDEBUG`.

mchesser, 2023-06-29:
I came across another use case for dynamic validation. Many of the shell interfaces use `device_get_binding` with an argument parsed from the shell console (e.g., [pwm](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/pwm/pwm_shell.c#L41), [gpio](https://github.com/zephyrproject-rtos/zephyr/blob/2f5068fcda0cb33284b6a8d777fbb95c9703ad82/drivers/gpio/gpio_shell.c#L63), [i2c](https://github.com/zephyrproject-rtos/zephyr/blob/2f5068fcda0cb33284b6a8d777fbb95c9703ad82/drivers/i2c/i2c_shell.c#L56) among others). Without any kind of validation, it's pretty easy to enter the wrong device and crash Zephyr or corrupt random state.

It would be good if there was some way to do typed device lookups to allow the shell to return an error instead of just using an invalid device.

(I realize this is a bit of an old issue at this point, but as far as I can tell there still doesn't seem to be any convenient method to do this kind of validation).