zephyrproject-rtos/zephyr #24965: RF2XX radio driver does automatic retransmission and OpenThread as well

markus-becker-tridonic-com, 2020-05-05:
The RF2XX radio driver zephyr/drivers/ieee802154/ieee802154_rf2xx.c does automatic retransmission for packets that have not been acknowledged (`RF2XX_TRX_PHY_STATE_CMD_TX_ARET_ON`). The default value from http://ww1.microchip.com/downloads/en/devicedoc/atmel-8351-mcu_wireless-at86rf233_datasheet.pdf of MAX_FRAME_RETRIES is 0x3.

In ieee802154_cc13xx_cc26xx.c the configurable define CONFIG_NET_L2_IEEE802154_RADIO_TX_RETRIES is used.
In ieee802154_mcr20a.c a hard-coded MCR20A_GET_SEQ_STATE_RETRIES is used.

A few questions:
* Should CONFIG_NET_L2_IEEE802154_RADIO_TX_RETRIES be used in all radio drivers?
* Should it only be used in subsys/net/l2/ieee802154/ieee802154_radio_csma_ca.c and not in the radio drivers? Should the radio drivers/hardware not do any retransmissions?
* Or if the radio driver/hardware does retransmissions, should subsys/net/l2/ieee802154/ieee802154_radio_csma_ca.c not do any of the retransmissions?
* Should CONFIG_NET_L2_IEEE802154_RADIO_TX_RETRIES be used to fuel the defines for OpenThread (`OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT`/`OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_INDIRECT`)?

@jukkar @tbursztyka @nandojve

rlubos, 2020-05-05:
Hey @markus-becker-tridonic-com, my 2 cents,

> Should CONFIG_NET_L2_IEEE802154_RADIO_TX_RETRIES be used in all radio drivers?

I don't think we should use this config in radio drivers since it's only available with 802.15.4 L2. Radio drivers should be L2 agnostic, as they can be used with different L2s (like we do with OpenThread for instance). 

> Should it only be used in subsys/net/l2/ieee802154/ieee802154_radio_csma_ca.c and not in the radio drivers? Should the radio drivers/hardware not do any retransmissions?

Yes, IMO L2 configs should only be used in the dedicated L2 implementation. Whether the radio driver is able to handle retransmissions or not is specified by `IEEE802154_HW_CSMA` capability.

> Or if the radio driver/hardware does retransmissions, should subsys/net/l2/ieee802154/ieee802154_radio_csma_ca.c not do any of the retransmissions?

This is actually how it works right now, if `IEEE802154_HW_CSMA` capability is returned by the radio driver, the 802.15.4 L2 will not use software CSMA/CA implementation (a thus retransmit):
https://github.com/zephyrproject-rtos/zephyr/blob/108820aafbc8cfb71abf5fbb6a97b81aed4bf1b5/subsys/net/l2/ieee802154/ieee802154.c#L278

> Should CONFIG_NET_L2_IEEE802154_RADIO_TX_RETRIES be used to fuel the defines for OpenThread (OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT/OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_INDIRECT)?

No, OpenThread is a different L2, we should not mix those.

markus-becker-tridonic-com, 2020-05-06:
> Yes, IMO L2 configs should only be used in the dedicated L2 implementation. Whether the radio driver is able to handle retransmissions or not is specified by `IEEE802154_HW_CSMA` capability.

The rf2xx driver seems to return the IEEE802154_HW_CSMA capability.

Is it then required to set OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT/OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_INDIRECT to 1 for OpenThread? Could/should this be automatically set for IEEE802154_HW_CSMA capable radio drivers?

Or should the radio driver be configurable to disable the automatic retries?

rlubos, 2020-05-06:
`IEEE802154_HW_CSMA` capability is forwarded to OpenThread (which defines its own caps):
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/openthread/platform/radio.c#L556
Based on those OT capabilities, the OT stack decides whether to handle CSMA on its own or not:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/openthread/platform/radio.c#L556

Hence, `OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT` etc should not be used when radio reports CSMA capability.

markus-becker-tridonic-com, 2020-05-06:
That capability was introduced in https://github.com/zephyrproject-rtos/zephyr/commit/108820aafbc8cfb71abf5fbb6a97b81aed4bf1b5, which we have not in our codebase yet. We will update. Thanks for the help.

closed 2020-05-06 by markus-becker-tridonic-com