zephyrproject-rtos/zephyr #24978: RFC: use compatible name for prefix for device-specific API

pabigot, 2020-05-05:
A number of devices including `ams,ccs811` and `maxim,ds3231` have capabilities that are not supported by generic subsystem APIs like sensor or counter.  Access to these capabilities is provided by device-specific extensions like:

```

int ccs811_envdata_update(struct device *dev,
                          const struct sensor_value *temperature,
                          const struct sensor_value *humidity);
int maxim_ds3231_synchronize(struct device *dev,
￼			     struct sys_notify *notify);
```

Zephyr does not have a global documented policy on naming that takes into account future conflicts, e.g. by placing things in namespaces.  This has recently caused several issues between external libraries and internal and public Zephyr code, and as Zephyr's in-tree capabilities expand the potential for conflict will only increase.

I propose that public API related to device-specific capabilities should use the device compatible for a prefix, e.g. `ams_ccs811_` and `maxim_ds3231_`.  This has several benefits:
* The longer name reduces the chance of conflict if two vendors produce devices with similar names;
* The vendor prefix will group the custom API in documentation indexes.
 
I don't propose that API for existing drivers like CCS811 be renamed (though I would support that), but only that new drivers like DS3231 (#17631) follow this practice.

carlescufi, 2020-05-05:
I couldn't agree more @pabigot . Adding this to today's API meeting.

carlescufi, 2020-05-05:
API meeting:
General consensus towards enforcing (and documenting) this. #24741 will include a patch to document this.

PR #24741 (2020-05-20): doc: expand on driver-specific extensions and usermode support for async APIs
Extend the provided template with annotated examples of API based on whether the functions must be invokable from user mode threads, since there are currently no in-tree examples of the specific techniques required.

Explain why `sys_notify` is not suitable for use in syscall API and identify the necessary replacement.

Closes #24978
closed 2020-05-20 by carlescufi