zephyrproject-rtos/zephyr #25012: checkpatch.pl doesn't match the vendor string properly

alexandru-porosanu-nxp, 2020-05-06:
**Describe the bug**
checkpatch.pl will complain that the vendor string is not documented, although this is not the case
Due to the slight difference between the vendor-prefixes.yaml from Linux and vendor-prefix

**To Reproduce**
Steps to reproduce the behavior:
1. ./scripts/checkpatch.pl -f ./dts/arm/nxp/nxp_kw41z.dtsi
2. Notice the WARNING:UNDOCUMENTED_DT_STRING: warnings for all the vendors that are in the specific checked file.
But looking into the vendor-prefixes.txt file, you can see both  ARM as well as NXP

**Expected behavior**
No warning reported

**Impact**
Annoyance

**Screenshots or console output**
``
dts/arm/nxp/nxp_kw41z.dtsi:40: WARNING:UNDOCUMENTED_DT_STRING: DT compatible string vendor "arm" appears un-documented -- check ./dts/bindings/vendor-prefixes.txt
#40: FILE: dts/arm/nxp/nxp_kw41z.dtsi:40:
+			compatible = "arm,cortex-m0+";

dts/arm/nxp/nxp_kw41z.dtsi:52: WARNING:UNDOCUMENTED_DT_STRING: DT compatible string "nxp,kw41z-mcg" appears un-documented -- check ./dts/bindings/
#52: FILE: dts/arm/nxp/nxp_kw41z.dtsi:52:
+			compatible = "nxp,kw41z-mcg";

dts/arm/nxp/nxp_kw41z.dtsi:52: WARNING:UNDOCUMENTED_DT_STRING: DT compatible string vendor "nxp" appears un-documented -- check ./dts/bindings/vendor-prefixes.txt
...
``
**Environment (please complete the following information):**
 - OS: Ubuntu 18.04 LTS
 - Toolchain: Zephyr SDK
 - Commit SHA: 2d9baf97f9fbda8cdd3eb05a82b505333c196cf9

**Additional context**
This issue seems to stem from the different format for vendor-prefixes.txt in Zephyr versus vendor-prefixes.yaml in Linux. The commit 5b10fac97efdb444b49f5e3c3ac16048ce8661c6 inadvertently replaces the check with the one from Linux.

stephanosio, 2020-05-06:
There is already a PR that fixes this:

alexandru-porosanu-nxp, 2020-05-06:
> There is already a PR that fixes this:
> #24959

Right, I missed that. I'll close this one (and the corresponding PR)

closed 2020-05-06 by carlescufi