zephyrproject-rtos/zephyr #25015: Bluetooth Isochronous Channels Support

DeltaEvo, 2020-05-06:
**Describe the solution you'd like**
Support Bluetooth LE 5.2 Isochronous Channel in zephyr controller subsystem

**Additional context**
Isochronous Channel is a new addition in Bluetooth 5.2 specification it the fondation for LE Audio

Is there any support planned ?

carlescufi, 2020-05-06:
@DeltaEvo yes, this is an ongoing effort that is currently mostly taking place in a private repository because it's implementing profiles that are not publicly released yet. We will start backporting to master soon, the parts that can be made public.
Access to the private repo is reserved for Bluetooth SIG Associate members that are also members of the Zephyr project, since there are legal limitations to sharing of the code.

DeltaEvo, 2020-05-06:
Will the backporting effort include Isochronous Channels ?

carlescufi, 2020-05-06:
> Will the backporting effort include Isochronous Channels ?

Well, yes, this is mainly what is being developed in the private tree. The rest of functionality is pushed directly to master in this repository.

closed 2020-05-06 by carlescufi

govindadas, 2020-10-06:
Any news on Isochronous Channels support in the Controller? There is (LE Audio) Host code in the Zephyr 2.4 release but it is not supported by the controller. 

> this is an ongoing effort that is currently mostly taking place in a private repository because it's implementing profiles that are not publicly released yet.

Is it possible to decuple the controller from the profiles? Isochronous Channels in the controller are important to me but I'm not a Bluetooth member.

govindadas, 2021-06-11:
@carlescufi any news yet?

carlescufi, 2021-06-11:
> @carlescufi any news yet?

Yes, the current open source controller (and Host) already has basic support for both BIS and CIS. Please try it out and report issues as you find them.

govindadas, 2021-06-11:
> > @carlescufi any news yet?
> 
> Yes, the current open source controller (and Host) already has basic support for both BIS and CIS. Please try it out and report issues as you find them.

Thanks! I will try it out.

bartmoons, 2021-08-11:
I have tried the `iso_receive` and `iso_broadcast` sample on both the `nrf52dk_nrf52832` and the `nrf53dk_nrf5340`. I'm unable to start scanning on both platforms, I get:
```
*** Booting Zephyr OS build zephyr-v2.6.0-1623-g997bbcb000c3  ***
Starting Synchronized Receiver Demo
...
Start scanning... failed (err -5)
...
[00:00:00.514,495] <wrn> bt_hci_core: opcode 0x2041 status 0x01
```
Does the example require some configuration which I'm unaware of?