zephyrproject-rtos/zephyr #25021: Problems getting open62541 to run on Zephyr

puchm, 2020-05-06:
Hi,

I am currently trying to get [open62541](https://github.com/open62541/open62541) running on Zephyr. The problem is that I am quite new to all of this, namely Zephyr, cmake and all of the infrastructure around C (e.g. how including libraries works).
**I have highlighted (bold) my questions for you since I didn't know which template I should use for these questions.**

## My setup
I am using the latest Zephyr version, 2.2.99.
My `app` folder is located at `~/zephyrproject/zephyr/app` and the folder structure is as follows:
```
├── CMakeLists.txt
├── prj.conf
└── src
    ├── build
    ├── main.c
    ├── open62541.c
    └── open62541.h
```
I generated the `open62541.c/h` files from the open62541 repository (master branch) following their guidelines for building.
The `app/CMakeLists.txt` contains the following:
```
cmake_minimum_required(VERSION 3.13.1)
find_package(Zephyr)
project(opc_ua_implementation)

target_sources(app PRIVATE src/open62541.c src/open62541.h src/main.c)
```
In `main.c` I am including open62541 using `#include "open62541.h"`. **Is this the right way and the right file to import or should I import `open62541.c`?**

I am building using `west build -b qemu_x86 app`, but later I want to switch from the emulation to a real board.

## Problems
The problem is that I am getting a number of errors, part of which I could fix and part of which I couldn't fix.
**One problem** is that open62541 is not finding a few files, which are:
- `sys/socket.h`
- `arpa/inet.h`
- `netinet/in.h`
- `netdb.h`
- `sys/ioctl.h`
I searched for those files and found them inside `zephyr/include/posix`. I added `posix/` to all of the include statements for these files inside `open62541.h` which solved that. **Is this a good thing to do or is there a better way?**

There is also a file which I can't fix that error for which is called `sys/select.h`. I searched the whole `zephyr` directory and didn't find the file anywhere. **What can I do about that?**

**Another problem** is the following:
`zephyrproject/zephyr/include/net/buf.h:785:17: error: 'CONFIG_NET_BUF_USER_DATA_SIZE' undeclared here (not in a function)`

I found out that this has to do with `Kconfig` so I tried around with that. I added `CONFIG_NET_BUF_USER_DATA_SIZE=10` to `prj.conf` and also tried it with a few other values but none of that worked. **Where else would I define that value so that it gets found by the compiler?**

I hope that there is someone who knows how to help me with this.
I would also suggest to make a forum or something like that for people who don't want to work on Zephyr but just need help asking questions since the issue templates suggest that this is actually not the place to ask for help.

Thank you so much for your help.
-Moritz

carlescufi, 2020-05-06:
I remember a while back Vincent Lacroix from Systerel presented integration of OPC UA at the TSC, but the work never got further as far as I can tell.
You can see his presentation here:
https://drive.google.com/file/d/1FZslmtc48I2U0S9JgmmMHTkPu-wRLN3r/view

cc @vla3913 assuming that is the right person. If not, perhaps they can redirect us to the appropriate person.

carlescufi, 2020-05-06:
In any case @puchm what I'd recommend is that you submit a [request for a module](https://docs.zephyrproject.org/latest/guides/modules.html) to include open62541 as part of the Zephyr set of repos. That way we will make sure we integrate with CI.

vla3913, 2020-05-06:
Hello @carlescufi.
Sorry we didn't keep you in touch.
We have successfully ported S2OPC (https://www.s2opc.com/) on Zephyr. At least, the client/server part is running on imx1064 and we are finishing the module to submit the result to the Zephyr project.
Current development for the porting activity can be found on this branch: https://gitlab.com/systerel/S2OPC/-/tree/plu_poc_zephyr_imx6.
Maybe, we will need the community support for last details. Module submission is a matter of days.

puchm, 2020-05-06:
Thank you @vla3913, that is really good to know.
Do you think an imx1020 would be sufficient to run a simple OPC UA server?
And is there already a guide on how to install the current S2OPC on Zephyr without having to wait for the module to be submitted?

jukkar, 2020-05-06:
@pfalcon didn't you work with open62541 at some point last year?

vla3913, 2020-05-06:
@puchm, I think an imx1020 is sufficient to run an OPC UA server, but if you activate data encryption and/or signature, you will have some extra-delay during the session activation.
I am going to open a specific issue on the s2opc module submission, i am going to cc you with the procedure.

vla3913, 2020-05-06:
Request for module creation submitted (see #25046).

pfalcon, 2020-05-07:
> @pfalcon didn't you work with open62541 at some point last year?

Yes, I made a proof of concept port about a year ago, it's available at https://github.com/pfalcon/open62541demos &  https://github.com/pfalcon/open62541 . "Proof of concept" is the keyword here. From my PoV, the biggest outcome of it was noticeable improvements to Zephyr POSIX subsys (but IIRC, not all improvements were merged, due to resistance POSIX subsys improvement faced).

vla3913, 2020-05-31:
I have difficulties to have the Mbedtls hardware entropy poll working on qemu x86. Do you know who I can contact on the subject ?

carlescufi, 2020-07-06:
See #25046

closed 2020-09-20 by github-actions[bot]