zephyrproject-rtos/zephyr #25022: There is no case’s information in serial log for ARC(R) HS Development Kit after one case was been flashed into the board.

xuhao8210, 2020-05-06:
**Describe the bug**
After one case was been flashed into the board:” ARC(R) HS Development Kit”, there is no case’s information in serial log, and only display the commit ID info. 
For example , after the hello world case was been flashed into the board, the serial log is only display :” *** Booting Zephyr OS build zephyr-v2.2.0-2375-gfdd85d5ad703  ***”. 

And I also checked that this problem was been happened after the commit ID: ” 8e653fe6b1aaa142c1f4f4526a3e8caecc522f20” checked in ,and its last commit ID:” f700022a35a1bb1d5eff94dca5942cfe0109164b” can work well. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b hsdk -p auto samples/hello_world/
2. west flash
3. check the serial log

**Expected behavior**
After the hello world case was been flashed into the board, the serial log should display below.

         *** Booting Zephyr OS build zephyr-v2.2.0-275-gf700022a35a1  ***
         Hello World! hsdk

**Screenshots or console output**

*** Booting Zephyr OS build zephyr-v2.2.0-276-g8e653fe6b1aa  ***
*** Booting Zephyr OS build zephyr-v2.2.0-276-g8e653fe6b1aa  ***
*** Booting Zephyr OS build zephyr-v2.2.0-275-gf700022a35a1  ***
Hello World! hsdk
*** Booting Zephyr OS build zephyr-v2.2.0-2375-gfdd85d5ad703  ***

**Environment (please complete the following information):**
 - OS:  ubuntu 18.04
 - ZEPHYR SDK  :   zephyr-sdk-0.11.2
 - Commit ID:  fdd85d5ad7034ff49d204818b4e0e02a86e96e70

xuhao8210, 2020-05-09:
After I ran the cases: "samples/synchronization/" , "samples/hello_world/","tests/kernel/queue/" and "tests/kernel/mem_protect/sys_sem/" , they are not all display the case’s information in serial log.
But the cases:"tests/kernel/pipe/pipe" and "tests/kernel/timer/timer_api/" can display the case’s information in serial log.

vonhust, 2020-05-11:
The root cause of this issue is incomplete support of arc multicore in openocd.  Sometimes, the current arc open cannot reset the hsdk status correctly, e.g. cache flush.  We are working on a fix of this in openocd,

When the metaware debugger is used to flash/debug the hsdk,  we can get the expected results:

```
*** Booting Zephyr OS build zephyr-v2.2.0-2719-ge59ea072e29e***
Hello World! hsdk
```

I suggest not to debug/flash hsdk before we fix the issues in openocd.

abrodkin, 2020-05-12:
From my previous experience that means that master core is infinitely waiting for slaves to wake up which never happens (note we have HSDK configured for SMP mode but in reality run stuff as if it's a normal single-core target).

To fix that one needs to run SMP-aware runner, which might be the one out of:
1. OpenOCD with SMP support for ARC. This is included in Zephyr's SDK starting from v0.11.3-beta-2 (you may get it here https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.11.3-beta-2)
2. MDB runner. See PR here https://github.com/zephyrproject-rtos/zephyr/pull/24806 and corresponding branch is here: https://github.com/foss-for-synopsys-dwc-arc-processors/zephyr/tree/topic-mdb-runner

Let me know if either option helps you to get any further.

xuhao8210, 2020-06-05:
After I used sanitycheck script to set up daily test for the ARC(R) HS Development Kit platform, I found that there are about 412 failure test cases on commit ID: 76d8f8f1e2, and all failure test case are same root cause that they don’t display the test result in serial log, so I updated this issue’s priority to high.  And please refer my sanitycheck command below for our daily test. 

Sanitycheck command :

sanitycheck -p hsdk --device-testing --device-serial /dev/ttyUSB0 -T  $HOME/zephyrproject/zephyr/tests/

nashif, 2020-06-05:
this is not a high prio bug, the issue is explained above. issues that are related to tools and individual boards are not considered high priority.

abrodkin, 2020-06-05:
So with our recent experiments the only solution to this problem is board reset in between tests. And here I mean not only separate `sanitycheck` invocations but each and every test-case.

Obviously that might be done by toggling a power source but:
1. It requires controllable power supply/strip
2. Integration of that power strip/supply control into `sanitycheck` infrastructure so that separate boards could be reset independently
3. Might require a bit longer time for safe board proper recovery, i.e. turn power off, wait ~10 seconds (to make sure all AC-DC capacitors are drained and the board is really off now), turn the power on.

In case of majority of ARC boards (excluding just a legacy EM StarterKit) we luckily have global board's reset wired to the same FTDI IC which is used for both JTAG connection & UART. And what we do internally we use a simple homebrew `rff-reset` utility (see https://github.com/foss-for-synopsys-dwc-arc-processors/rff-ftdi-reset, kudos to @Palmyr3) to cold-reset our boards.

What we may do is to integrate its utilization by the `sanitycheck` script, for example we tried it that way:
```diff
--- a/scripts/sanity_chk/sanitylib.py
+++ b/scripts/sanity_chk/sanitylib.py
@@ -545,6 +545,10 @@ class DeviceHandler(Handler):
 
     def handle(self):
         out_state = "failed"
+        os.system('rff-reset --serial 251642000087')
+        time.sleep(5)
 
         if self.suite.west_flash:
             command = ["west", "flash", "--skip-rebuild", "-d", self.build_dir]
```

Indeed that's just a mock-up and we should do it more-elegant and still quite universal because upcoming MDB-runner (https://github.com/zephyrproject-rtos/zephyr/pull/24806) will also benefit from this feature.

So if my proposal with reset makes sense let's discuss how to do it in the most elegant way.

Also if we decide to go that way we may want to package `rff-reset` in the SDK. That we may do super-easy via `meta-zephyr-sdk` or in any other way given simplicity of the utility and not that many dependencies.

xuhao8210, 2020-06-12:
I checked that the newest SDK with version: zephyr-sdk-0.11.3 have included the solution(OpenOCD with SMP support for ARC) from the tag : v0.11.3-beta-2, but the problem is still exist after I tried the newest source code with commit ID: 8cd6bb6e61b4b17cdc30ada76d7b8a188d4f93ad.
BTW, as I have no license for the MDB runner, I have not tried the solution.

abrodkin, 2020-06-17:
@xuhao8210 so there're more things to try:
1. Run HSDK in SMP mode in OpenOCD for real, for this you'll need this https://github.com/zephyrproject-rtos/zephyr/pull/25955.
2. Even with true SMP execution we noticed the board may enter non-recoverable [via JTAG] state and the only way to recover is to reset the board. Fortunately it could be done w/o any extra HW via the same USB wire which is used for both serial port & JTAG. The only thing which is needed is our tiny utility (https://github.com/foss-for-synopsys-dwc-arc-processors/rff-ftdi-reset).

So that's now an interesting discussion on if we may bundle it into Zephyr's SDK or tools (whatever you call it). Given we have it in sources and it gets built with a trivial Makefile it might be integrated in whatever build system we use in Zephyr. Would be good to hear back from @nashif, @galak & @stephanosio on that one.

xuhao8210, 2020-07-02:
@abrodkin  Thanks for your support. After I tried your solution that I added below row in the script:” scripts/sanity_chk/sanitylib.py” to test the newest commit ID for Zephyr Repo, I found that some test cases(e.g. tests/portability/ and  tests/lib/ ) are still failure as they reset the board failure, could you please help to check them ?

============add below code in the file:” sanitylib.py” ==================
         os.system('rff-reset --serial 251642000171')
        time.sleep(10)
        os.system('sudo chmod 666 /dev/ttyUSB0')
        time.sleep(10)
========================================================
BTW, if I did not add the row :”os.system('sudo chmod 666 /dev/ttyUSB0')”, the script was also report error that it can’t open the serial port. And it seems that each platform will be reset for your solution, whether we should find a better way to only reset the hsdk platform, so what are your comments ?

abrodkin, 2020-07-03:
@xuhao8210 could you please provide more details on how do those tests still fail?
As for `chmod` on `/dev/ttyUSB0` I'm pretty sure it's not needed at all, instead please add your local user to the group which has access to serial ports. Typically it's a `dialout` group. It could be done that way on Ubuntu:
```
sudo usermod -a -G dialout $USER
```

xuhao8210, 2020-07-06:
@abrodkin  For the test case:” tests/portability” on Commit ID: 897b5bf60a, could you please check its log below ?  BTW, it seems that each platform will be reset for your solution(updating sanitylib.py file), whether we should find a better way to only reset the hsdk platform, so what are your comments ?

==================Test log for tests/portability/ case=====================
_[_ztest@zserver13 zephyr]$ scripts/sanitycheck -p hsdk --device-testing --device-serial /dev/ttyUSB0 -T tests/portability/ -vv
Renaming output directory to /home/ztest/work/zephyrproject/zephyr/sanity-out.5
INFO    - JOBS: 4
DEBUG   - Reading test case configuration files under /home/ztest/work/zephyrproject/zephyr/tests/portability...
DEBUG   - scanning /home/ztest/work/zephyrproject/zephyr/tests/portability
DEBUG   - scanning /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v1
DEBUG   - Found possible test case in /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v1
DEBUG   - scanning /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2
DEBUG   - Found possible test case in /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2
DEBUG   - Reading platform configuration files under /home/ztest/work/zephyrproject/zephyr/boards...
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/gpmrb/gpmrb.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/minnowboard/minnowboard.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/acrn/acrn.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/up_squared/up_squared.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/up_squared/up_squared_32.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86_nommu.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86_coverage.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86_64.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/posix/native_posix/native_posix.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/posix/native_posix/native_posix_64.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/posix/nrf52_bsim/nrf52_bsim.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp_em6.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp_em7d_esp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp_em9d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp_em7d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp_em5d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/emsdp/emsdp_em4.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/nsim/nsim_hs_smp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/nsim/nsim_hs.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/nsim/nsim_em.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/nsim/nsim_sem.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/nsim/nsim_sem_mpu_stack_guard.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/em_starterkit/em_starterkit_em7d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/em_starterkit/em_starterkit.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/em_starterkit/em_starterkit_em7d_v22.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/em_starterkit/em_starterkit_em11d.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/hsdk/hsdk.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/qemu_arc/qemu_arc_em.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/qemu_arc/qemu_arc_hs.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/arc/iotdk/iotdk.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_s1000_crb/intel_s1000_crb.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/xtensa/qemu_xtensa/qemu_xtensa.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/up_squared_adsp.yaml
DEBUG   - Found platform configuration /home/ztest/work/zephyrproject/zephyr/boards/xtensa/esp32/esp32.yaml
[... cut ...]
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/sanity-out/hsdk/tests/portability/cmsis_rtos_v2/portability.cmsis_rtos_v2/handler.log

INFO    - 1 of 2 tests passed (50.00%), 1 failed, 0 skipped with 0 warnings in 61.21 seconds
INFO    - In total 2 test cases were executed on 1 out of total 260 platforms (0.38%)
INFO    - 2 tests executed on platforms, 0 tests were only built.

Hardware distribution summary:

| Board   | ID   |   Counter |
|---------|------|-----------|
| hsdk    |      |         2 |__

============================================================

abrodkin, 2020-07-06:
@xuhao8210 OK so this needs to be finally solved and I'll take a look at it seriously this time around, see https://github.com/zephyrproject-rtos/zephyr/issues/26665 :)

xuhao8210, 2020-07-07:
@abrodkin  OK, Got it . Thanks

closed 2020-08-31 by carlescufi

abrodkin, 2020-09-07:
@xuhao8210 with https://github.com/zephyrproject-rtos/zephyr/pull/27793 merged we now do reset in between tests that way:
```
echo "#!/bin/bash" > pre_script_file.sh
echo "rff-reset --serial ${DIGILENT_SERIAL_NUMBER} && sleep 2" >> pre_script_file.sh
chmod a+x pre_script_file.sh
./scripts/sanitycheck ... --pre-script pre_script_file.sh ...
```
And with that we're getting much better results on the HSDK board.
Please let us know if it helps on your side.

P.S. Sources and build instructions for `rff-reset` utility are available here: https://github.com/foss-for-synopsys-dwc-arc-processors/rff-ftdi-reset