zephyrproject-rtos/zephyr #25032: build failure

wentongwu, 2020-05-06:
**Describe the bug**
```
[78/105] Building C object zephyr/boards/arm/lpcxpresso55s16/CMakeFiles/boards__arm__lpcxpresso55s16.dir/pinmux.c.obj
FAILED: zephyr/boards/arm/lpcxpresso55s16/CMakeFiles/boards__arm__lpcxpresso55s16.dir/pinmux.c.obj 
ccache /opt/sdk/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE=CONFIG_FLASH_SIZE -DBUILD_VERSION=zephyr-v2.2.0-2390-g99493ab4a882 -DCPU_LPC55S16JBD100 -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../../../../../../../drivers -I../../../../../../../include -Izephyr/include/generated -I../../../../../../../soc/arm/nxp_lpc/lpc55xxx -I/home/buildslave/src/github.com/zephyrproject-rtos/modules/hal/cmsis/CMSIS/Core/Include -I/home/buildslave/src/github.com/zephyrproject-rtos/modules/hal/nxp/mcux/devices/LPC55S16 -I/home/buildslave/src/github.com/zephyrproject-rtos/modules/hal/nxp/mcux/drivers/lpc/. -isystem ../../../../../../../lib/libc/minimal/include -isystem /opt/sdk/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /opt/sdk/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/sanity-out/lpcxpresso55s16_ns/samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m33 -mthumb -mabi=aapcs -imacros/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/samples/portability/cmsis_rtos_v2/timer_synchronization=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/buildslave/src/github.com/zephyrproject-rtos/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/buildslave/src/github.com/zephyrproject-rtos=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/boards/arm/lpcxpresso55s16/CMakeFiles/boards__arm__lpcxpresso55s16.dir/pinmux.c.obj -MF zephyr/boards/arm/lpcxpresso55s16/CMakeFiles/boards__arm__lpcxpresso55s16.dir/pinmux.c.obj.d -o zephyr/boards/arm/lpcxpresso55s16/CMakeFiles/boards__arm__lpcxpresso55s16.dir/pinmux.c.obj   -c /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/boards/arm/lpcxpresso55s16/pinmux.c
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/boards/arm/lpcxpresso55s16/pinmux.c: In function 'lpcxpresso_55s16_pinmux_init':
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/boards/arm/lpcxpresso55s16/pinmux.c:121:16: error: missing binary operator before token "("
  121 | #if DT_HAS_NODE(DT_NODELABEL(flexcomm4)) && \
      |                ^
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/boards/arm/lpcxpresso55s16/pinmux.c:150:16: error: missing binary operator before token "("
  150 | #if DT_HAS_NODE(DT_NODELABEL(hs_lspi))
      |                ^
[79/105] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mem_slab.c.obj
```

more info here https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/71697/summary/console

wentongwu, 2020-05-06:
This issue at least blocks my PR https://github.com/zephyrproject-rtos/zephyr/pull/22904

carlescufi, 2020-05-06:
This was fixed by #25020

closed 2020-05-06 by carlescufi