zephyrproject-rtos/zephyr #25036: k_pipe: available read / write space helper functions

cfriedt, 2020-05-06:
**Is your enhancement proposal related to a problem? Please describe.**
It's not immediately obvious how much of a k_pipe buffer is available for reading / writing (if the buffer even exists - they can be created with no buffer).

**Describe the solution you'd like**
Just a few extra functions. I've already written the latter two already (for buffered pipes), but I'm sure some improvements might be in order to also address unbuffered pipes.
```
bool k_pipe_has_buffer(const struct k_pipe *pipe);
size_t k_pipe_read_avail(const struct k_pipe *pipe);
size_t k_pipe_write_avail(const struct k_pipe *pipe);
```

**Describe alternatives you've considered**
I didn't see any other alternatives, so I just wrote them myself. 

**Additional context**
Was going to keep them static inside of my socketpair(2) implementation (#24813) but was asked to make a separate PR for this by @pfalcon CC @carlescufi .

pfalcon, 2020-05-06:
+1 for adding this functionality, the usecase is pretty clear.

cfriedt, 2020-05-06:
Also, for reasons that are probably obvious to the maintainers, I'm going to add a timeout parameter as well; even though the pipes are constant, it's possible that another thread could be modifying values concurrently and therefore it would probably be safe to hold the semaphore.

cfriedt, 2020-05-06:
CC @pfalcon @carlescufi

PR #25038 (2020-05-07): kernel: pipe: read_avail / write_avail syscalls
These helper functions may clarify how much data is available to read
from or write to the circular buffer of a k_pipe.

Fixes #25036
commit 3315f8fe (2020-05-07): kernel: pipe: read_avail / write_avail syscalls

These helper functions may clarify how much data is available to read
from or write to the circular buffer of a k_pipe.

Fixes #25036
closed 2020-05-07 by carlescufi