zephyrproject-rtos/zephyr #25051: tests/drivers/gpio/gpio_api_1pin failed on reel_board and mec15xxevb_assy6853.

chen-png, 2020-05-07:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto  tests/drivers/gpio/gpio_api_1pin 
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-2414-gcbd9c86b5ce2  ***
Running test suite gpio_api_1pin_test

**starting test - test_gpio_pin_configure_push_pull
Running test on port=GPIO_0, pin=11
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_config.c:31: pin_get_raw_and_verify: (val_expected not equal to val_actual)
Test point 10: invalid pin get value
FAIL - test_gpio_pin_configure_push_pull**

starting test - test_gpio_pin_configure_single_ended
Running test on port=GPIO_0, pin=11
When configured as input test pin value is high
PASS - test_gpio_pin_configure_single_ended

starting test - test_gpio_pin_set_get_raw
Running test on port=GPIO_0, pin=11
PASS - test_gpio_pin_set_get_raw

starting test - test_gpio_pin_set_get
....

starting test - test_gpio_int_edge_both
Running test on port=GPIO_0, pin=11
Pin interrupt is not supported.
SKIP - test_gpio_int_edge_both

starting test - test_gpio_int_edge_to_active
Step 1: Configure pin as active high
Running test on port=GPIO_0, pin=11
Pin interrupt is not supported.
SKIP - test_gpio_int_edge_to_active

starting test - test_gpio_int_edge_to_inactive
Step 1: Configure pin as active high
Running test on port=GPIO_0, pin=11
Pin interrupt is not supported.
SKIP - test_gpio_int_edge_to_inactive

starting test - test_gpio_int_level_high
Running test on port=GPIO_0, pin=11
PASS - test_gpio_int_level_high

starting test - test_gpio_int_level_low
Running test on port=GPIO_0, pin=11
PASS - test_gpio_int_level_low

starting test - test_gpio_int_level_active
Step 1: Configure pin as active high
Running test on port=GPIO_0, pin=11
Step 2: Configure pin as active low
Running test on port=GPIO_0, pin=11
PASS - test_gpio_int_level_active

starting test - test_gpio_int_level_inactive
Step 1: Configure pin as active high
Running test on port=GPIO_0, pin=11
Step 2: Configure pin as active low
Running test on port=GPIO_0, pin=11
PASS - test_gpio_int_level_inactive

starting test - test_gpio_pin_toggle_visual
Running test on port=GPIO_0, pin=11
LED ON
LED OFF
LED ON
LED OFF
PASS - test_gpio_pin_toggle_visual

Test suite gpio_api_1pin_test failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.2
 - Commit ID: cbd9c86b5c

dcpleung, 2020-05-08:
On mec15xxevb_assy6853, are there jumpers on `JP31 13-14` (for `GPIO156`) and `JP41 1-2`? These two have pull-up effect which will make the pin high if set as input.

chen-png, 2020-05-12:
@dcpleung  i removed the JP31 13-14 and JP41 1-2,  **the test_gpio_pin_configure_push_pull passed, but it failed at test_gpio_int_edge_falling.**

starting test - test_gpio_int_edge_falling
DEBUG   - DEVICE: Running test on port=GPIO140_176, pin=14
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_pin_interrupt.c:103: test_gpio_pin_interrupt_edge: **(cb_count not equal to cb_count_expected)
DEBUG   - DEVICE: Test point 0: Pin interrupt triggered invalid number of times on rising/to active edge**
DEBUG   - DEVICE: FAIL - test_gpio_int_edge_falling

dcpleung, 2020-05-12:
The interrupt tests should be fixed by #25155

chen-png, 2020-05-13:
this test also failed on sam_e70_xplained and up_squared.

stephanosio, 2020-05-13:
@chen-png Are you sure this test failed on `sam_e70_xplained`? I just ran sanitycheck on it and did not see the failure.

`gpio_api_2pin` (not `gpio_api_1pin`) fails, but that is only because I did not make the necessary testing connections on the board.

chen-png, 2020-05-13:
> @chen-png Are you sure this test failed on `sam_e70_xplained`? I just ran sanitycheck on it and did not see the failure.

for this test tests/drivers/gpio/gpio_api_1pin, do i need to do any gpio connections for sam_e70_xplained?

this is the test result with no gpio connections.

*** Booting Zephyr OS build v2.3.0-rc1-20-g9a37154151d5  ***
Running test suite gpio_api_1pin_test

starting test - test_gpio_pin_configure_push_pull
Running test on port=PORTC, pin=8
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_config.c:31: pin_get_raw_and_verify: (val_expected not equal to val_actual)
Test point 10: invalid pin get value
FAIL - test_gpio_pin_configure_push_pull

stephanosio, 2020-05-13:
@chen-png No GPIO connection necessary; this test passes for me on `sam_e70_xplained`:
```
*** Booting Zephyr OS build v2.3.0-rc1-26-g4980a13fc4f7  ***
Running test suite gpio_api_1pin_test
===================================================================
starting test - test_gpio_pin_configure_push_pull
Running test on port=PORTC, pin=8
PASS - test_gpio_pin_configure_push_pull
===================================================================
starting test - test_gpio_pin_configure_single_ended
Running test on port=PORTC, pin=8
When configured as input test pin value is high
Open Drain configuration or Pull Up pin bias is not supported
SKIP - test_gpio_pin_configure_single_ended
===================================================================
starting test - test_gpio_pin_set_get_raw
Running test on port=PORTC, pin=8
PASS - test_gpio_pin_set_get_raw
===================================================================
starting test - test_gpio_pin_set_get
Running test on port=PORTC, pin=8
PASS - test_gpio_pin_set_get
===================================================================
starting test - test_gpio_pin_set_get_active_high
Running test on port=PORTC, pin=8
Step 1: Set logical, get logical and physical pin value
Step 2: Set physical, get logical and physical pin value
PASS - test_gpio_pin_set_get_active_high
===================================================================
starting test - test_gpio_pin_set_get_active_low
Running test on port=PORTC, pin=8
Step 1: Set logical, get logical and physical pin value
Step 2: Set physical, get logical and physical pin value
PASS - test_gpio_pin_set_get_active_low
===================================================================
starting test - test_gpio_pin_toggle
Running test on port=PORTC, pin=8
PASS - test_gpio_pin_toggle
===================================================================
starting test - test_gpio_port_set_masked_get_raw
Running test on port=PORTC, pin=8
PASS - test_gpio_port_set_masked_get_raw
===================================================================
starting test - test_gpio_port_set_masked_get
Running test on port=PORTC, pin=8
PASS - test_gpio_port_set_masked_get
===================================================================
starting test - test_gpio_port_set_masked_get_active_high
Running test on port=PORTC, pin=8
Step 1: Set logical, get logical and physical port value
Step 2: Set physical, get logical and physical port value
PASS - test_gpio_port_set_masked_get_active_high
===================================================================
starting test - test_gpio_port_set_masked_get_active_low
Running test on port=PORTC, pin=8
Step 1: Set logical, get logical and physical port value
Step 2: Set physical, get logical and physical port value
PASS - test_gpio_port_set_masked_get_active_low
===================================================================
starting test - test_gpio_port_set_bits_clear_bits_raw
Running test on port=PORTC, pin=8
PASS - test_gpio_port_set_bits_clear_bits_raw
===================================================================
starting test - test_gpio_port_set_bits_clear_bits
Running test on port=PORTC, pin=8
PASS - test_gpio_port_set_bits_clear_bits
===================================================================
starting test - test_gpio_port_set_clr_bits_raw
Running test on port=PORTC, pin=8
PASS - test_gpio_port_set_clr_bits_raw
[... 44 lines cut ...]
Step 1: Configure pin as active high
Running test on port=PORTC, pin=8
Step 2: Configure pin as active low
Running test on port=PORTC, pin=8
PASS - test_gpio_int_level_active
===================================================================
starting test - test_gpio_int_level_inactive
Step 1: Configure pin as active high
Running test on port=PORTC, pin=8
Step 2: Configure pin as active low
Running test on port=PORTC, pin=8
PASS - test_gpio_int_level_inactive
===================================================================
starting test - test_gpio_pin_toggle_visual
Running test on port=PORTC, pin=8
LED ON
LED OFF
LED ON
LED OFF
PASS - test_gpio_pin_toggle_visual
===================================================================
Test suite gpio_api_1pin_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

chen-png, 2020-05-21:
>  No GPIO connection necessary; this test passes for me on `sam_e70_xplained`:

the test is passed on sam_e70_xplained, maybe it's caused by my testing environment before.

chen-png, 2020-05-21:
> this test also failed on sam_e70_xplained and up_squared.

for up_squared, this test is passed now due to #25384

chen-png, 2020-05-21:
this test is passed on mec15xxevb_assy6853 now

closed 2020-08-04 by github-actions[bot]