zephyrproject-rtos/zephyr #25055: Redundant flash shell commands

nordic-krch, 2020-05-07:
**Describe the bug**
It's rather "bad impression" bug. `samples/drivers/flash_shell` implements commands to handle flash and there are already generic commands in `drivers/flash/flash_shell.c` which are not enabled in that example (because they overlap). It would be good to merge those two and clean up the example or even remove it and just enable flash commands in generic shell example.

nordic-krch, 2020-05-07:
@nvlsianpu fyi

eugmes, 2020-09-29:
It seems that something has changed since this issue was reported. Right now I see two entries for 'flash' when I run the `flash_shell` sample. `CONFIG_FLASH_SHELL` is enabled for that sample.

erwango, 2022-11-30:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/14260.
Closing this one as the other one is the most commented

closed 2022-11-30 by erwango