zephyrproject-rtos/zephyr #25084: LLDP: missing net_pkt_set_lldp in lldp_send

ehudmarvell, 2020-05-07:
Sending lldp packet includes ARP field in its eth->type instead LLDP type. ( wireshark is attached)

[lldp.zip](https://github.com/zephyrproject-rtos/zephyr/files/4593709/lldp.zip)

Steps to reproduce the behavior:
compile with CONFIG_NET_LLDP=y
add to ethernet driver :
1. capabilities: ETHERNET_LLDP
2. net_lldp_set_lldpdu(iface);

I would expect that the driver will send lldp packet instead I see that the packet which should be lldp
packet includes in its ethernet header ptype=ARP(the rest of the packet seem fine).

I think it because net_pkt_set_lldp(pkt, true); is missing from lldp_send function.
As result ethernet_send not inserted to 
> } else if (IS_ENABLED(CONFIG_NET_LLDP) && net_pkt_is_lldp(pkt)) {
> 		ptype = htons(NET_ETH_PTYPE_LLDP);

and it inserted to instead
> 	} else if (IS_ENABLED(CONFIG_NET_ARP)) {
> 		/* Unktown type: Unqueued pkt is an ARP reply.
> 		 */
> 		ptype = htons(NET_ETH_PTYPE_ARP);
> 		net_pkt_set_family(pkt, AF_INET);
> 	}

**Environment (please complete the following information):**
 -  Zephyr SDK
 - Commit SHA :6933248e0cb4f7af31e2bab5b39c594806ab53ac

What do you think?

Thanks,
Ehud

jukkar, 2020-05-07:
Yep, your analysis is correct. The fix is easy, I will send a fix shortly.

jukkar, 2020-05-07:
After the fix, the Ethernet type is correct. Unfortunately wireshark shows now that the packet is malformed.

ehudmarvell, 2020-05-07:
@jukkar thank you, I don't see any problem after the fix in my wireshark

PR #25086 (2020-05-08): net: lldp: Sent LLDP packet was missing proper net_pkt type
The LLDP packet was created but its type was not set to LLDP
and was sent as ARP message.

Fixes #25084
commit ad9f32a4 (2020-05-08): net: lldp: Sent LLDP packet was missing proper net_pkt type

The LLDP packet was created but its type was not set to LLDP
and was sent as ARP message.

Fixes #25084
closed 2020-05-08 by jukkar