zephyrproject-rtos/zephyr #25093: optimize 32-bit x86 system calls

andrewboie, 2020-05-07:
**Is your enhancement proposal related to a problem? Please describe.**
System calls on 32-bit x86 use the old-style "int 0x80" mechanism to trap to kernel mode. There are more efficient ways to do this.

**Describe the solution you'd like**
Use SYSENTER/SYSEXIT instead to perform syscalls.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

andrewboie, 2020-08-27:
Top two commits of this old branch is where I started working on this, but I set it aside. https://github.com/andrewboie/zephyr/tree/sysenterexit

dcpleung, 2024-11-07:
`SYSENTER` is not very secure as it does not save `EIP` and `ESP` automatically. This needs to be done with code in ring 3. `SYSENTER` just jumps to syscall entry and has to trust whatever coming `EIP` and `ESP` are. This can be abused to return to some arbitrary code after handling the syscall. So not going to implement this.

closed 2024-11-07 by dcpleung