zephyrproject-rtos/zephyr #25098: MCUX I2C bus errors leave state machine in busy state

marketideas, 2020-05-07:
**Describe the bug**
A bus error on the I2C bus will cause the I2C state machine to stay in a busy state and all future I2C transactions will fail.

**To Reproduce**
I2C external devices can become 'stuck' waiting to finalize a bit shift clock while driving the open-collector SDA line low.  This prevents the master from being able to drive the SDA line until the I2C bus is cleared.
1) Perform a CPU hard reset during an I2C read transaction.  You have to get a bit lucky here to have it happen during an inbound byte transfer. The device will get stuck driving SDA and waiting on additional clocks from the CPU that will never come.
2) do a shell i2c scan command to verify that no devices are present.
3) It is now impossible to get the I2C devices back without a power off.

**Expected behavior**
I2C bus should be able to be reset without power off and/or hard/soft reset.  Bus clearing requires sendings 9 clock signals on SCL to clear all devices from driving the bus.  Then, the second bug is within Zephyr, in that previous transactions have failed leaving the I2C state machine in a busy state, so no new transactions may begin.

**Impact**
Critical software can not recover without a physical power off.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Unbuntu 18.03
 - Toolchain: zephyr SDK
 - Commit: will update once commit has a pull request.

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-05-07:
This is being addressed in:
https://github.com/zephyrproject-rtos/zephyr/pull/23442

carlescufi, 2020-05-12:
@marketideas I think the right course of action here is likely to implement `i2c_recover_bus()` for the mcux driver, perhaps in a manner similar to the one you suggest [here](https://github.com/zephyrproject-rtos/zephyr/pull/23442#issuecomment-625845557).

henrikbrixandersen, 2020-05-12:
> @marketideas I think the right course of action here is likely to implement `i2c_recover_bus()` for the mcux driver, perhaps in a manner similar to the one you suggest [here](https://github.com/zephyrproject-rtos/zephyr/pull/23442#issuecomment-625845557).

I disagree. Those are two, separate problems. This PR deals with fixing the internal state machine _if_ the I2S bus transfer fails.

henrikbrixandersen, 2020-06-23:
@marketideas Which board is this on? We have several "MCUX I2C" drivers in the tree.

henrikbrixandersen, 2020-07-06:
Assuming this is related to #25096.

PR #26761 (2020-07-09): drivers: i2c: mcux: return to idle on failed transfers
Abort I2C transfers through the MCUX HAL if a transfer fails to avoid deadlock in the driver/HAL.

Fixes #25098
commit 99c75978 (2020-07-09): drivers: i2c: mcux: return to idle on failed transfers

Abort I2C transfers through the MCUX HAL if a transfer fails to avoid
deadlock in the driver/HAL.

Fixes #25098
closed 2020-07-09 by MaureenHelm