zephyrproject-rtos/zephyr #25109: Flash tests fail on posix

sglass68, 2020-05-07:
**Describe the bug**
test_erase and test_align fail in the flash_simulator test suite

**To Reproduce**
Steps to reproduce the behavior:
1. west build --force -b native_posix tests/drivers/flash_simulator
2. build/zephyr/zephyr.elf
3. See error

**Expected behavior**
All tests should pass

**Impact**
Causes confusion

**Screenshots or console output**
build/zephyr/zephyr.elf 
*** Booting Zephyr OS build zephyr-v2.2.0-2512-gb358ca5e522d  ***
Running test suite flash_sim_api
===================================================================
starting test - test_int
PASS - test_int
===================================================================
starting test - test_write_read
PASS - test_write_read
===================================================================
starting test - test_erase
Incremental pattern expected
Erased area expected
Incremental pattern expected

    Assertion failed at tests/drivers/flash_simulator/src/main.c:62: test_check_pattern32: (val32 not equal to r_val32)
flash word at offset 0x2000 has value 0x80002
FAIL - test_erase
===================================================================
starting test - test_access
PASS - test_access
===================================================================
starting test - test_out_of_bounds
PASS - test_out_of_bounds
===================================================================
starting test - test_align

    Assertion failed at tests/drivers/flash_simulator/src/main.c:230: test_align: (-EINVAL not equal to rc)
Unexpected error code (0)
FAIL - test_align
===================================================================
starting test - test_double_write
PASS - test_double_write
===================================================================
Test suite flash_sim_api failed.
===================================================================
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Lniux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: 61c6e2b172

**Additional context**
Add any other context about the problem here.

vanwinkeljan, 2020-05-08:
Not a bug either an `enhancement`  as `flash_simulator` tests are not supported by `native posix` (only `qemu_x86` is whitelisted)

closed 2020-05-09 by nashif

sglass68, 2020-05-09:
Should we change the tests to support native_posix? It seems excessive to spin up qemu for a such a simple test. If so, how?

sglass68, 2020-05-09:
OK I see you have added support, but it can only run once. How to fix that?

galak, 2020-05-09:
> OK I see you have added support, but it can only run once. How to fix that?

what do you mean run once?

sglass68, 2020-05-09:
The first time I run it, it creates flash.bin and passes.

The second time I run it, it uses the old flash.bin and fails. Updated log:

/build/zephyr/zephyr.elf 
*** Booting Zephyr OS build zephyr-v2.2.0-2692-gff856e01f5d8  ***
Running test suite flash_sim_api
===================================================================
starting test - test_int

    Assertion failed at tests/drivers/flash_simulator/src/main.c:84: test_int: (0xff not equal to test_read_buf[i])
sim flash byte at offset 0x0 has value 0x002
FAIL - test_int
===================================================================
starting test - test_write_read

    Assertion failed at tests/drivers/flash_simulator/src/main.c:102: test_write_read: (0 not equal to rc)
flash_write (-5) should succedd at off 0x0
FAIL - test_write_read
===================================================================
starting test - test_erase
Incremental pattern expected

    Assertion failed at tests/drivers/flash_simulator/src/main.c:62: test_check_pattern32: (val32 not equal to r_val32)
flash word at offset 0x4 has value 0xffffffff02
FAIL - test_erase
===================================================================
starting test - test_access
PASS - test_access
===================================================================
starting test - test_out_of_bounds
PASS - test_out_of_bounds
===================================================================
starting test - test_align
PASS - test_align
===================================================================
starting test - test_double_write
PASS - test_double_write
===================================================================
Test suite flash_sim_api failed.
===================================================================
PROJECT EXECUTION FAILED

galak, 2020-05-09:
> The first time I run it, it creates flash.bin and passes.
> 
> The second time I run it, it uses the old flash.bin and fails. Updated log:

Gotcha, we should open a new issue for that bug.

sglass68, 2020-05-09:
OK, added
https://github.com/zephyrproject-rtos/zephyr/issues/25171

commit 57671d97 (2020-05-09): tests: flash_simulator: Add native posix support

Added support for native posix boards to flash_simulator tests by
making sure that flash layout lines up with layout expected by
tests.

Resolves #25109