zephyrproject-rtos/zephyr #25128: Missing `python3-dev` dependency

dieselburner, 2020-05-08:
**Describe the bug**

Failure to install Python dependencies due to missing `python3-dev` package. Package is not mentioned in [Install dependencies](https://docs.zephyrproject.org/latest/getting_started/index.html#install-dependencies).

**To Reproduce**

1. Install Ubuntu 20.04 LTS on a machine.
2. Follow [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html#getting-started-guide)
3. Fail on last step of [Get Zephyr and install Python dependencies](https://docs.zephyrproject.org/latest/getting_started/index.html#get-zephyr-and-install-python-dependencies)

**Expected behavior**

No errors after `pip3 install --user -r ~/zephyrproject/zephyr/scripts/requirements.txt`.

**Impact**

Annoyance. Solved via `sudo apt install python3-dev`.

**Screenshots or console output**

```
  ...
  psutil/_psutil_common.c:9:10: fatal error: Python.h: No such file or directory
      9 | #include <Python.h>
        |          ^~~~~~~~~~
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for psutil1
  ...
```

**Environment (please complete the following information):**

 - OS: Ubuntu 20.04 LTS
 - Toolchain: N/A
 - Commit SHA: `7778c22b6031ae766df7cd08522eb9675a0aeb69`

carlescufi, 2020-05-08:
@dieselburner would you mind sending a PR for this?

dieselburner, 2020-05-08:
> @dieselburner would you mind sending a PR for this?

Yes, give me a moment.

dieselburner, 2020-05-08:
@carlescufi here you go.

carlescufi, 2020-05-08:
> @carlescufi here you go.

Thanks!

PR #25129 (2020-05-08): doc: gsg: add missing python3-dev dependency
Linux distro might not have a python3-dev package installed by default,
which will give an error during Python dependencies installation.

Closes #25128.
commit 02a5ca93 (2020-05-08): doc: gsg: add missing python3-dev dependency

Linux distro might not have a python3-dev package installed by default,
which will give an error during Python dependencies installation.

Closes #25128.
closed 2020-05-08 by carlescufi