zephyrproject-rtos/zephyr #25140: Unable to obtain dhcp lease

dieselburner, 2020-05-08:
**Describe the bug**

Unable to obtain DHCPv4 lease.

**To Reproduce**

1. Build and flash `samples/net/dhcpv4_client` app.
2. Continuously execute `net iface` command in DUT shell.

**Expected behavior**

DHCPv4 lease obtained and DUT gets it's own IP address.

**Impact**

Showstopper

**Screenshots or console output**

*samples/net/dhcpv4_client*:

```
uart:~$ net iface

Interface 0x20010aa0 (Ethernet) [1]
===================================
Link addr : 02:80:E1:00:27:59
MTU       : 1500
Ethernet capabilities supported:
        10 Mbits
        100 Mbits
IPv4 unicast addresses (max 1):
        <none>
IPv4 multicast addresses (max 1):
        <none>
DHCPv4 lease time : 0
DHCPv4 renew time : 0
DHCPv4 server     : 0.0.0.0
DHCPv4 requested  : 0.0.0.0
DHCPv4 state      : selecting
DHCPv4 attempts   : 7
```

*samples/net/cloud/google_iot_mqtt*:

```
*** Booting Zephyr OS build zephyr-v2.2.0-2422-g8e5b8059501e  ***
[00:00:01.595,000] <inf> net_google_iot_mqtt: Main entered
[00:00:01.595,000] <inf> net_google_iot_mqtt: starting DHCPv4
<freeze>
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: `edeb555a8b3bd2e680a9fdf2de56a06b763a6c56`

**Additional context**

DUT : `nucleo_f756zg`

dieselburner, 2020-05-08:
And yes, I'm 95% sure it's not my environment. Will try to bisect the faulty commit or see if there's really something with my network. But I have more than 10 devices in a network right now, and all of them seems to work.

dieselburner, 2020-05-08:
Bug confirmed. Output from v2.2.0:

```
uart:~$ net iface

Interface 0x20017ba0 (Ethernet) [1]
===================================
Link addr : 02:80:E1:00:46:E8
MTU       : 1500
Ethernet capabilities supported:
        10 Mbits
        100 Mbits
IPv4 unicast addresses (max 1):
        192.168.0.191 DHCP preferred
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.168.0.1
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 7200
DHCPv4 renew time : 3600
DHCPv4 server     : 192.168.0.1
DHCPv4 requested  : 192.168.0.191
DHCPv4 state      : bound
DHCPv4 attempts   : 1
[00:00:08.987,000] <inf> net_dhcpv4: Received: 192.168.0.191
[00:00:08.987,000] <inf> net_dhcpv4_client_sample: Your address: 192.168.0.191
[00:00:08.987,000] <inf> net_dhcpv4_client_sample: Lease time: 7200 seconds
[00:00:08.987,000] <inf> net_dhcpv4_client_sample: Subnet: 255.255.255.0
[00:00:08.987,000] <inf> net_dhcpv4_client_sample: Router: 192.168.0.1
```

jukkar, 2020-05-08:
DHCPv4 works just fine:
```
[00:00:03.120,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.120,000] <inf> net_echo_server_sample: Network connected
[00:00:03.120,000] <inf> net_echo_server_sample: Starting...
[00:00:03.120,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.120,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:03.120,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:03.120,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv4)...
[00:00:05.160,000] <inf> net_dhcpv4: Received: 192.0.2.24
[00:00:05.160,000] <inf> net_config: IPv4 address: 192.0.2.24
[00:00:05.160,000] <inf> net_config: Lease time: 3600 seconds
[00:00:05.160,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:05.160,000] <inf> net_config: Router: 192.0.2.2
uart:~$ net iface

Interface 0x0807af20 (Ethernet) [1]
===================================
Link addr : 00:00:5E:00:53:3B
MTU       : 1500
Ethernet capabilities supported:
        Virtual LAN
IPv6 unicast addresses (max 3):
        fe80::200:5eff:fe00:533b autoconf preferred infinite
        2001:db8::1 manual preferred infinite
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ff00:533b
        ff02::1:ff00:1
IPv6 prefixes (max 2):
        <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 43300
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
        192.0.2.24 DHCP preferred
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.0.2.2
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 3600
DHCPv4 renew time : 1800
DHCPv4 server     : 192.0.2.2
DHCPv4 requested  : 192.0.2.24
DHCPv4 state      : bound
DHCPv4 attempts   : 1
```
Just to make sure, I also tried with dhcpv4_client sample:
```
[00:00:02.160,000] <inf> net_dhcpv4: Received: 192.0.2.24
[00:00:02.160,000] <inf> net_dhcpv4_client_sample: Your address: 192.0.2.24
[00:00:02.160,000] <inf> net_dhcpv4_client_sample: Lease time: 3600 seconds
[00:00:02.160,000] <inf> net_dhcpv4_client_sample: Subnet: 255.255.255.0
[00:00:02.160,000] <inf> net_dhcpv4_client_sample: Router: 192.0.2.2
uart:~$ net iface

Interface 0x080672c0 (Ethernet) [1]
===================================
Link addr : 00:00:5E:00:53:3B
MTU       : 1500
Ethernet capabilities supported:
        Virtual LAN
IPv4 unicast addresses (max 1):
        192.0.2.24 DHCP preferred
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.0.2.2
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 3600
DHCPv4 renew time : 1800
DHCPv4 server     : 192.0.2.2
DHCPv4 requested  : 192.0.2.24
DHCPv4 state      : bound
DHCPv4 attempts   : 1
uart:~$ 
```
This with native_posix. Note that yesterday there was qemu issue fixed that affected networking, make sure you use very latest master.

jukkar, 2020-05-08:
After checking your git commit, I can confirm that you were affected with this issue which was reverted after you had tried it `x86: Revert "qemu_x86: use icount for 32-bit" d2dce5e845abf4965cee130ac4dacb4fa370aa38`
Closing this one as this is already fixed.

dieselburner, 2020-05-08:
Problem still present.

jukkar, 2020-06-09:
> Nope. Issue is still there.

Lets re-open the issue then.

dieselburner, 2020-06-09:
Thanks. I will come back with the results after trying out again on latest master.

MaureenHelm, 2020-06-16:
@dieselburner is this issue still in master?

dieselburner, 2020-06-16:
Sorry, borrowed my dev board to another person. Will retest as soon as I have it, in a day or two.

dieselburner, 2020-06-28:
Tested on the latest tip, and it works again.

```
*** Booting Zephyr OS build zephyr-v2.3.0-636-gf4896adaa86c  ***
[00:00:01.610,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:06.137,000] <inf> net_dhcpv4: Received: 192.168.0.106
[00:00:06.137,000] <inf> net_dhcpv4_client_sample: Your address: 192.168.0.106
[00:00:06.137,000] <inf> net_dhcpv4_client_sample: Lease time: 86400 seconds
[00:00:06.137,000] <inf> net_dhcpv4_client_sample: Subnet: 255.255.255.0
[00:00:06.137,000] <inf> net_dhcpv4_client_sample: Router: 192.168.0.1
```

Closing.

closed 2020-06-28 by dieselburner

MaureenHelm, 2020-06-29:
@dieselburner thanks for following up