zephyrproject-rtos/zephyr #25153: Support for buildtesting SoCs w/o boards

galak, 2020-05-08:
We don't have any particular means today to test SoC support w/o a board.

Starting this thread to use for discussion on if and how we might support this.

My initial thought is that we'd just create a fake board port for the SoC, have some policy to limit this to one board per SoC "family variant" so we're not having an explosion.

Having some naming convention like `boards/<arch>/buildtest_<SOC>`.

galak, 2020-05-14:
Dev-Review [May 14]:
* @MaureenHelm - will look into what ref board may exist from a NXP point of view
* @pabigot - what is the bar w/regards to maintenance acceptance for such code/SoC code.
* @nashif - we should always have a board to build the SoC code.  How do support incubation for code that might not be ready for whatever reasons.  A staging concept.

Look at doing w/modules, need a skeleton for SoC and board module to show how to do this.  (@carlescufi there's a PR from @tejlmand to make it easier to support out of tree SoCs & boards - wait on getting that merged).

MaureenHelm, 2020-05-14:
> * @MaureenHelm - will look into what ref board may exist from a NXP point of view

The recommended board for this SoC is MAPS-KS22, you can see this in the MCUXpresso SDK builder on the right side:

[image]

I'm not sure if this board is available anymore. Even if it is, the MCUXpresso SDK is an old version with the Clear BSD license.

tejlmand, 2020-07-09:
> Look at doing w/modules, need a skeleton for SoC and board module to show how to do this. (@carlescufi there's a PR from @tejlmand to make it easier to support out of tree SoCs & boards - wait on getting that merged).

https://github.com/zephyrproject-rtos/zephyr/pull/26715 created.

tejlmand, 2025-01-15:
closing.
Support for SoC and boards in Zephyr modules was done in https://github.com/zephyrproject-rtos/zephyr/pull/26715.

In-tree SoCs should always have a board, see https://github.com/zephyrproject-rtos/zephyr/issues/25153#issuecomment-628702744
> @nashif - we should always have a board to build the SoC code. How do support incubation for code that might not be ready for whatever reasons. A staging concept.

The staging concept is achieved by the ability to create / develop and test the SoC in a Zephyr module, and when a corresponding board, which can be upstreamed, is available, the SoC and board can be contributed upstream.

closed 2025-01-15 by tejlmand